CircuitSetup/Split-Single-Phase-Energy-Meter #3: Using with single phase UK (240v) mains power

motechsolutions, 2019-08-26:
What changes would i have to make to the firmware (Arduino sketch files) to allow the split phase energy meter to correctly monitor UK 240v single phase mains power.
Im using a 9V AC 240V transformer type power supply and two CT current sensors (SCT-013-000 100A/50mA)

Any advise would be much appreciated

motechsolutions, 2019-08-26:
Hi
Yes im using the latest version from your repo :)
The EMONESP web interface shows V1 as 122v and V2 as 122v and the total voltage shows 244v, is this correct (voltage gain field = 4485). The strange thing is if i set the freq in the web interface or in energy_meter.h to 389 (50hz value) then the total voltage value is the same as V1 (122v). how to i get the v1 value to display the correct voltage (ie 244v) as the "Voltage gain" value will not go above 65535 and ive calculated i need to have a value of  (real v=244v / v1=122v * V-Gain = 38730) 77460

Thank you in advance

motechsolutions, 2019-08-27:
Hi again
That worked a treat, Voltage now shows correct 243v. one quick thing, what do i need to change to get the following setup for watts data.
Watt1
Watt2
Total Watts (Watt1)
my setup is a home with a office. a single phase 240v mains power to house which also supplies power to the office at the back, (Single power distribution board with separate circuit for office)

Thanks in advance

motechsolutions, 2019-08-27:
That sorted the EMONESP web interface, :)

I think for me it may be a better to display and track the two CT sensors independently on EMONESP and EMONCMS , make CT1 and CT2  independent (your thoughts ?)

Values in EMONESP Web and values sent to EMONCMS

**Circuit1:**
CT1
W1
PF1

Circuit2
CT2
W2
PF2

I  think energy_meter.cpp/h and circuitsetup_split-phase.json would need to be changed but i may be wrong (newbie to Arduino etc)

Thanks in advance

Also ive uploaded my split phase energy monitor to ESP32 adapter board to easyeda if anyone wants to use it for a 30pin ESP32S & 0.96" OLED display
https://easyeda.com/motechsolutions/Split-Phase-Energy-Monitor-ESP32-Adapter

motechsolutions, 2019-08-27:
I think ive sorted circuitsetup_split-phase.json

`{
    "name": "Single Phase Energy Meter",
    "category": "CircuitSetup",
    "group": "Energy Monitoring",
    "description": "CircuitSetup Split Single Phase Energy Meter",
    "inputs": [
		{
            "name": "CT1",
            "description": "Current CT1",
            "processList": [
                {
                    "process": "log_to_feed",
                    "arguments": {"type": "ProcessArg::FEEDID", "value": "Current CT1" }
                }
            ]
        },
		{
            "name": "CT2",
            "description": "Current CT2",
            "processList": [
                {
                    "process": "log_to_feed",
                    "arguments": {"type": "ProcessArg::FEEDID", "value": "Current CT2" }
                }
            ]
        },
        {
            "name": "PF",
            "description": "Power Factor",
            "processList": [
                {
                    "process": "log_to_feed",
                    "arguments": {"type": "ProcessArg::FEEDID", "value": "Power Factor" }
                }
            ]
        },
		{
            "name": "temp",
            "description": "Temp",
            "processList": [
                {
                    "process": "log_to_feed",
                    "arguments": {"type": "ProcessArg::FEEDID", "value": "Temp" }
                }
            ]
        },
		{
            "name": "totV",
            "description": "Voltage",
            "processList": [
                {
                    "process": "log_to_feed",
                    "arguments": {"type": "ProcessArg::FEEDID", "value": "Voltage" }
                }
            ]
        },
		{
            "name": "W2",
            "description": "Watts CT2",
            "processList": [
                {
                    "process": "log_to_feed",
                    "arguments": {"type": "ProcessArg::FEEDID", "value": "Watts CT2" }
                }
            ]
        },
        {
            "name": "W1",
            "description": "Watts CT1",
            "processList": [
                {
                    "process": "log_to_feed",
                    "arguments": {"type": "ProcessArg::FEEDID", "value": "Watts CT1" }
                },
                {
                    "process": "power_to_kwh",
                    "arguments": {"type": "ProcessArg::FEEDID", "value": "CT1 kWh" }
                }
            ]
        }
    ],

    "feeds": [
        {
            "name": "Current CT1",
            "type": "DataType::REALTIME",
            "engine": "Engine::PHPFINA",
            "interval": "10",
            "unit": "A"
        },
        {
            "name": "Current CT2",
            "type": "DataType::REALTIME",
            "engine": "Engine::PHPFINA",
            "interval": "10",
            "unit": "A"
        },
        {
            "name": "Voltage",
            "type": "DataType::REALTIME",
            "engine": "Engine::PHPFINA",
            "interval": "10",
            "unit": "V"
        },
		{
            "name": "Watts CT1",
            "type": "DataType::REALTIME",
            "engine": "Engine::PHPFINA",
            "interval": "10",
            "unit": "W"
        },
		{
            "name": "Watts CT2",
            "type": "DataType::REALTIME",
            "engine": "Engine::PHPFINA",
            "interval": "10",
            "unit": "W"
        },
		{
            "name": "Power Factor",
            "type": "DataType::REALTIME",
            "engine": "Engine::PHPFINA",
            "interval": "10",
            "unit": ""
        },
		{
            "name": "Temp",
            "type": "DataType::REALTIME",
            "engine": "Engine::PHPFINA",
            "interval": "10",
            "unit": "°C"
        },
		{
            "name": "CT1 kWh",
            "type": "DataType::REALTIME",
            "engine": "Engine::PHPFINA",
            "interval": "10",
            "unit": "kWh"
        }
    ]
}
`

CircuitSetup, 2019-08-29:
You can edit the device json file, but you should be able to just change or add the feeds too. When they export from EmonESP, they should appear in EmonCMS anyway.

Thanks for sharing the EasyEDA board as well. That looks good! You may need some extra power for the display though. If wifi on the ESP32 has trouble staying connected and freezes, that may be why. I would recommend having a separate 3v3 input to power the diplay and/or board.

motechsolutions, 2019-08-29:
After reading this post (https://community.hiveeyes.org/t/power-supply-for-the-esp32/1859/7)
Ive added a 47uF tantalum capacitor and a 100uF low-ESR electrolytic capacitor to the 3.3v rail next to the ESP32 3.3v pin, on testing with a scope and DVM the current pull from the ESP32 and OLed is max 175ma and the 3.3v power rail looks more stable but i need to now soak test my design.
Even if i dont use the OLed display the extra capacitors can only help the ESP32 power rail.
Ive updated my EasyEDA design so yourself and others can see the new adapter PCB design
Your thoughts?

CircuitSetup, 2019-08-29:
Those caps will definitely help with the spikes. By default if a connection is lost, the esp will go into AP mode, which I've found uses the most power starting up, and then staying up. 

FYI, The energy meter board has a 22uf cap on the 3v3 output from the SMPS. 

If you need to reduce the power any more you can turn down the esp clock to 80mhz when EmonESP is compiled, or turn off going into AP mode when a connection is lost in web_server.cpp

closed 2019-09-09 by CircuitSetup

CircuitSetup, 2019-09-22:
Curious if you've got this up and running yet?

motechsolutions, 2019-09-23:
Hi John

Yes working great, the first CT monitors my house and the second CT monitors my office.
[image]

CircuitSetup, 2019-09-23:
Very nice! I'm sure some other Europeans would benefit from your calibration settings with that AC Transformer. Do you mind sharing?

motechsolutions, 2019-09-24:
No problem at all, 
**AC transformer** = https://www.ebay.co.uk/itm/9V-AC-AC-Adaptor-Transformer-Power-Supply-5-5mm-UK-Plug/312408988666?epid=1342924577&hash=item48bd06dbfa:g:eiIAAOSwVH9cKlL5

**Calibration Settings**
Voltage = 38990
CT1 & CT2 = 25758 
Freq = 389
CT Gain = 21