CircuitSetup/Split-Single-Phase-Energy-Meter #4: Issue with device crashing after x hours SSPEM

jreiners, 2019-09-17:
Not sure if this is the forum, and I doubt it is, and I'm sorry.

I've installed the split single phase energy monitor on my breaker box without issue, and configured wifi with good signal at -61db, but randomly and frequently since I've got it running a few hours it seems the monitor device locks up and drops off wifi.

The only way I know to get it back is to cause an unexpected power interruption, and suddenly everything seems to work for a bit.

I have not got it working with emoncms yet, it can see the inputs, and they are all green but I haven't actually seen it plot anything.

I assume it might be locking up, just wondering what I can do to troubleshoot.

The values I can see are below:

Key | Value
-- | --
V1 | 119.29V
V2 | 119.30V
CT1 | 22.7940A
CT2 | 22.7920A
totI | 45.5860A
W | 5005.5142W
PF | 0.05
temp | 27.00
freq | 60.02

Thanks for your help, and I am excited for this project.

CircuitSetup, 2019-09-18:
Glad you got things up and running, but sorry you're having connection issues! 
You have a kit, correct? 

In testing, if the wi-fi signal was low, and the ESP32 lost the connection, it would go into SoftAP mode, which uses more power. That repeating over and over would sometimes cause the brownout detector to trigger, and the ESP32 would lock. -70db is usually around where I would see this happen.  

Try loading [this firmware for EmonESP:](https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/raw/master/Software/EmonESP/src/emonesp_v2.5.2_noAP_lost_conn.bin)

If it loses the connection to the main AP, it will not go into SoftAP mode. Since it already has your router credentials stored, you don't need the SoftAP (unless you want to change which AP it connects to)

Also, does your AP happen to have the same name with 2.5ghz and 5ghz, or do you have a repeater with the same name? This could be confusing the ESP32 as well if it drops a connection and tries to connect to the other physical AP.

One last thing - your PF seems to be low. You may have your CT clamps on backwards, but it really doesn't matter if you don't care about that value, have solar panels, or want to use any of the other metering values.

CircuitSetup, 2019-09-18:
> can I possibly use a different, even higher voltage/current brick to power it?
If you have a higher amperage *AC* Transformer you can also try that. The one provided is 1A, but there are also 1.5A Jameco ones out there (along with others that I haven't personally tested). You'll probably have to calibrate for that transformer though. Higher voltage won't matter in this case.

jreiners, 2019-09-18:
yes I received my kit yesterday!!!

I'll try loading that firmware and let you know how it goes, it's offline again atm, I set it up temporarily on emoncms a couple hours ago, and it lasted about an hour before it dropped offline. Now that I am home I just checked to see if it was broadcasting its SSID and its not.

I have a pretty simple ubiquiti setup with a few access points spread around, none of the SSID are shared between 2.4 and 5ghz, I'm used to IOT wifi controllers that are 2.4ghz only so I have everything set for easy onboarding of devices.

I'll try the firmware first, I'd rather disable the ap anyways. If that doesn't work I might have another AC transformer in the shop.

As for PF, I noticed that and will fix, thanks for pointing that out,  I'd rather get it working for a few hours first.

CircuitSetup, 2019-09-18:
> yes I received my kit yesterday!!!

awesome! 

> I have a pretty simple ubiquiti setup with a few access points spread around, none of the SSID are shared between 2.4 and 5ghz, I'm used to IOT wifi controllers that are 2.4ghz only so I have everything set for easy onboarding of devices.

That shouldn't be causing any issues then.

> I'll try the firmware first, I'd rather disable the ap anyways. If that doesn't work I might have another AC transformer in the shop.

If you have one without the same plug as the existing, you can use the VC+ and GND inputs next to the jack on the board. Further, if you want to only give a voltage signal to the board while VC+ is connected, you can sever the JP3 trace on the back of the board, and power the rest of the electronics through the ESP32 with a phone charger. Just make sure it's not a crappy one or, if it is use a usb cable with a ferrite bead.

jreiners, 2019-09-18:
> If you have one without the same plug as the existing, you can use the VC+ and GND inputs next to the jack on the board. Further, if you want to only give a voltage signal to the board while VC+ is [...]

Very nice, when I went to reboot it I noticed the box had a hole that lined up with usb, I might go that route if it keeps dropping, that would probably take care of all of my issues. I have some fairly nice 2.4A 5v Samsung chargers. 

I updated to the above firmware before dinner, sadly I still had a drop about 20 minutes later.

I took the panel apart and flipped the backwards one, readings are better. I also rose the meter up higher above the panel instead of below, which increased my signal another 7 percent.

fingers crossed.

jreiners, 2019-09-18:
OK, After moving the device above the breaker panel, it worked all night without crashing.

Another question, after moving everything I noticed that the PF was low again, I have both clamps installed before my main breaker, with arrows pointing in the direction of my main breaker. Should I flip one around so they are opposite? It doesn't seem right.

Thank you for your help.

CircuitSetup, 2019-09-18:
Glad to hear it's staying connected after moving it a bit closer! It seems that some of these ESP32s use a tiny bit more power than others to stay connected.

How low is your PF getting? The CTs should be pointing in the same direction.

jreiners, 2019-09-18:
Here is my latest grab from the device itself. Thanks for the help!

I really don't have much worry about my PF is I'll get the correct W readings

Key | Value
-- | --
V1 | 119.17V
V2 | 119.20V
CT1 | 22.4450A
CT2 | 19.9800A
totI | 42.4250A
W | 4676.0234W
PF | -0.02
temp | 27.00
freq | 59.99

I did make sure each arrow was pointing toward the breaker

CircuitSetup, 2019-09-18:
Interesting, it's possible the arrow is reversed on the CT. If it doesn't matter to you, then it's okay :)

jreiners, 2019-09-18:
I did think of that, possibly the coil inside is reversed, as they both are in the orientation with cord leaving the top side. If it bothers me later I'll flip it around. Looks like I'm all set. Thanks again.

closed 2019-09-18 by jreiners

digiblur, 2019-09-18:
@jreiners I know in my area of the US when I changed code bases on my 2 channel meter it made me realize I should have the clamps opposite of one another in direction since the the lines were out of phase from each other.

jreiners, 2019-09-18:
That's probably it, I'll try switching it again, it crashed on me again a bit ago, I got about 12 hours before it died. 

I did try to desolder J3 last night, but no matter what I did it still powered up the ESP32, and lost my `second` voltage reading so I resoldered it.

Can I just flush clip +5v or GND to the ESP32 and then use the USB power or is +5v needed from the other board for anything?

CircuitSetup, 2019-09-18:
> @jreiners I know in my area of the US when I changed code bases on my 2 channel meter it made me realize I should have the clamps opposite of one another in direction since the the lines were out of phase from each other.

Really? The phase shouldn't matter - it's all about the direction of the winding on the coil. I wonder if one of your CTs is wired backwards internally.

> That's probably it, I'll try switching it again, it crashed on me again a bit ago, I got about 12 hours before it died.

I'm honestly not sure why that keeps happening with yours. I would say try putting in another ESP32 to see if that makes a difference, but there aren't many that are as narrow as the NodeMCU one. 

> I did try to desolder J3 last night, but no matter what I did it still powered up the ESP32, and lost my `second` voltage reading so I resoldered it.

Sorry, I should have mentioned you have to physically cut the jumper with a knife.

> Can I just flush clip +5v or GND to the ESP32 and then use the USB power or is +5v needed from the other board for anything?

If you power through the ESP32, (assuming JP3 is cut or the rectifier diode is removed) the rest of the electronics would be powered via the ESP32's 3V3 power supply (LDO). I wouldn't recommend hooking up both the transformer and ESP32 power at the same time for very long. If you hook up the 5v pin on the ESP32 it should still go through to the 3V3 output.

digiblur, 2019-09-18:
Yep. Definitely threw me at first. Amps were fine but wattage was negative. So I multiplied the value to flip it. Then later I changed the phase the power supply was on after installing a closer plug. Then the value flipped again. It seemed to follow the reference voltage phase.

jreiners, 2019-09-25:
I think I might have found something, I noticed my energy meter crashing may be caused by having it's web page open in multiple tabs, it happened again yesterday when I didn't realize I already had a tab open in chrome, and opened another one, about 20 minutes later my meter dropped off the network.

I use an app for work called sysdweb to control systemd services, and when multiple users logged it it would crash similarly, probably no fix for it, but figured I'd offer the info.

I tried it again this morning, and 12 minutes later it crashed on me, and I needed to reboot it.

I ended up buying an internet switch so I can bounce it from anywhere.

CircuitSetup, 2019-09-26:
This may actually make sense. If the interface is open itll refresh via async. In testing async was throwing random error messages that didnt make sense. This may have been causing a memory leak. 

The library for AsyncTCP was just updated a couple days ago to address an issue that looks related. I'll look into this more and release a patch.

jreiners, 2019-09-26:
Sweet! Hopefully it helps, that is the only way I can crash it. 

Is that something I'll need to compile and reflash?

CircuitSetup, 2019-09-26:
It would only happen if the interface was left open. Let me know if not having it open helps. Either way, I'll release a new firmware version with the AsyncTCP update.

jreiners, 2019-09-26:
Yeah, it works fine for days without issue if I don't open it, and leave it open refreshing, no need to really anyways if it's working.

Emoncms works fine open for days so I'm good.

Thanks for this.

CircuitSetup, 2019-09-26:
Glad to hear!

More specifically, this also may have been causing issues, and was just recently fixed: https://github.com/me-no-dev/ESPAsyncWebServer/commit/2f3703702987e31249d4c5c9d1f90cebf1ffa9e8

CircuitSetup, 2019-10-03:
> @jreiners I know in my area of the US when I changed code bases on my 2 channel meter it made me realize I should have the clamps opposite of one another in direction since the the lines were out of phase from each other.

I was just thinking about this again, and you are right. Since the voltage reference, by default, is only from one side of the panel (on the pcb the voltage signal is split unless the jp3 jumper is severed). Current from the other side will be out of phase from that voltage, and appear to the meter as a negative value. I don't know why that didn't click in my head sooner.