CircuitSetup/Split-Single-Phase-Energy-Meter #5: Errors compiling firmware (+suggested fix)

kobuki, 2019-09-25:
When compiling the EmonESP FW, there are errors compiling the project using the default build task. I had to make 2 changes to fix that:

* In web_server.cpp, delete the `#include <SPIFFS.h>` line. It's apparently unused and just causes problems.
* In platformio.ini, change `lib_deps` to `lib_deps = PubSubClient@2.6, ESP Async WebServer@1.1.1, ../libraries/ATM90E32` to automatically make use of the included library.

Is the code up to date in the repository? Maybe these small issues have already been fixed but not committed. Or maybe should I use the pr/2 branch?

CircuitSetup, 2019-09-25:
Thanks for the notes @kobuki ! 

I hadn't compiled everything with platformio. I'll make that change to the platform.ini file, and wifi.cpp

I have to look into the #include <SPIFFS.h> line. It should need that for the file system for the files that need to be uploaded in /data

Also, what MCU are you compiling for? There may be something different between versions for the ESP8266 and ESP32.

kobuki, 2019-09-25:
Right, ESP8266 as only that is defined as platform. Though it would be nice to have ESP32 configs, too. 

Compilation shows a bunch of warnings but it's eventually successful. SPIFFS is part of the standard Arduino framework now, and references such as SPIFFS are defined in `spiffs_api.cpp`, for instance, which does get compiled along with other related files. If it compiles in the Arduino IDE, I think it's safe to remove. I tend to avoid that IDE nowadays, though.

CircuitSetup, 2019-09-25:
I just added the esp32 config. Although I haven't tested it, everything should be there.

Good to know regarding SPIFFS, I'll test that, thanks!

kobuki, 2019-09-25:
I've just tested it in the Arduino IDE (v1.8.8) and The ESP8266 framework (v2.4.2) with a generic ESP8266 board. It only compiles if I remove/comment out that SPIFFS.h include line, so the situation is the same as for PIO. This is not needed - unless you compile using a very old ESP8266 framework where it might have been.

CircuitSetup, 2019-09-27:
Okay, I'll make that a condition of the ESP8266. I tested with an ESP32 and it still needs to include SPIFFS.h

closed 2019-09-27 by CircuitSetup