CircuitSetup/Split-Single-Phase-Energy-Meter #6: Using web interface and/or leaving it open causes ESP32 memory leak/crash

CircuitSetup, 2019-09-26:
I think I might have found something, I noticed my energy meter crashing may be caused by having it's web page open in multiple tabs, it happened again yesterday when I didn't realize I already had a tab open in chrome, and opened another one, about 20 minutes later my meter dropped off the network.

I use an app for work called sysdweb to control systemd services, and when multiple users logged it it would crash similarly, probably no fix for it, but figured I'd offer the info.

I tried it again this morning, and 12 minutes later it crashed on me, and I needed to reboot it.

I ended up buying an internet switch so I can bounce it from anywhere.

_Originally posted by @jreiners in https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/issues/4#issuecomment-535126441_

CircuitSetup, 2019-09-26:
Split this into a new issue. 

Recent updates to AsyncTCP: https://github.com/me-no-dev/AsyncTCP/pull/67
and ESPAsyncWebserver: https://github.com/me-no-dev/ESPAsyncWebServer/pull/591

Should solve this problem. Testing shortly...

CircuitSetup, 2019-09-26:
According to the ESPAsyncWebServer fix, it may also cause a crash if you:

- reset the ESP32
- visit the web interface with a Chrome based browser

Doesn't matter if you close the browser or not, the session wasn't being closed properly in all cases, which would result in a memory leak and eventually cause a crash.

jreiners, 2019-09-26:
Ahh nice catch, I always use Chrome.

CircuitSetup, 2019-09-27:
Okay, this is fixed with v2.5.3 https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/releases/tag/2.5.3

closed 2019-09-27 by CircuitSetup

jreiners, 2019-09-27:
Thanks, I'll check it out.

Justin Reiners
Reiners Cloud Consulting LLC
I'm happy to be your local Omaha distributor for 3CX and SimpleHelp!
Contact me for info.