CircuitSetup/Split-Single-Phase-Energy-Meter #7: Calibration setting for 240V regions

amil109, 2019-09-29:
Hi there,

I'm not sure if this is definitely the right place to submit this, but it is a software issue (I suppose) so I figured here was as good as anywhere!

I live in the UK, and have one of the Split-Single-Phase energy meters - it's working very well, except I need to be able to calibrate the voltage to read correctly.

I have a 230VAC/9VAC transformer, and get a completely sensible reading of ~122V in EmonESP. This corresponds to an actual line voltage of ~244V (measured with a multi-meter).

Using the calibration calculation in the readme, based on the default voltage gain:

VoltageGain = 37106 * 244 / 122 = 74212

When I try to put this value in the GUI, I get an error, telling me it's above 65535 (I suppose it's a 16bit integer).

Would it be possible to add a 'switch' for 240V operation? I can post-process the data in EmonCMS and can display the corrected value in OpenHab based on the MQTT feed, but it would be nice to be able to have it corrected at source.

Happy to write some code myself if you can point me to the right place!

Aside from the minor issue above, thanks for the great product!!!

Cheers,

amil109

CircuitSetup, 2019-09-29:
Make sure you change the frequency setting to 50hz in the config. After that is changed EmonESP should adjust for 220/240v.

netomx, 2019-10-02:
But what about 240V and 60Hz?

CircuitSetup, 2019-10-02:
> But what about 240V and 60Hz?

I actually didnt realize that this was used until recently. What country are you using the meter in? I'll get a fix out for this shortly.

netomx, 2019-10-02:
Mexico. Industries use 240V and 60Hz. Households use 127V

CircuitSetup, 2019-10-03:
Right. Is that still split single phase, or 2 phase? It shouldn't matter as long as you only have 2 mains wires to measure and it's not 3 phase.

netomx, 2019-10-03:
2 phases of 127v

amil109, 2019-10-04:
I can confirm that switching to 50Hz worked for me. 

I had put the calibration value only for frequency in the configuration before I posted, but it only seemed to switch voltages once I also put in a calibration value for the voltage itself. 

In any case, all's good now!

motechsolutions, 2019-10-15:
I had the same issue as im in the UK as well, Circuit Setup fixed the 240V problem a while back, take a look at https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/issues/3 to see the fixes i needed for 240v 50Hz uk

cfunseth, 2019-12-02:
Regarding calibration settings, I'm attempting to use the Home Assistant MQTT sketch provided in this repo but I'm having some issues with an incorrect voltage being reported.  I purchased the kit from CircuitSetup and everything was working out of the box correctly with EmonESP loaded.  When I flashed the example MQTT sketch though, I'm getting a voltage reading of ~590V on each phase.  

I'm wondering how the 37,106 value for "VoltageGain" was determined?  Assuming I'm on a 120V/60Hz system, I'd imagine there's a 13.3x ratio (120V/9V) as well as the voltage divider being used on the schematic, but that only looks to to account for another 11x ratio (110k/10k).  Therefore I'm getting a total gain of 146.67, but that results in a value that's way too low.  Is there something I'm missing from the Atmel IC?

BitKnitting, 2019-12-03:
IDK if this will help..but here is what I learned regarding calibration
<https://github.com/BitKnitting/FitHome/wiki/ElectricityMonitor#calibration>

CircuitSetup, 2019-12-03:
@cfunseth I believe I sent you a hand assembled version of the energy meter with resistors for the voltage divider that will come on the next batch (which are 20k and 10k). The calibration value for voltage can be changed to 7611 in the MQTT sketch.

closed 2019-12-03 by CircuitSetup

cfunseth, 2019-12-03:
@CircuitSetup Ah, thank you for clarifying!  Would you mind explaining how the 7,611 value was calculated?  I can't find anything in the ATM90E32 datasheet and even with the resistor changes you mentioned I still don't see where.  I didn't want to create a new issue on this but can if you'd prefer to keep this closed.