CircuitSetup/Split-Single-Phase-Energy-Meter #9: The fields sent from ESP don't match CircuitSetup defaults on SD card image

veggie2u, 2019-10-01:
The fields I am getting from the ESP via MQTT are as follows:
0, V1, V2, CT1, CT2, totl, W, SolarV, SCT1, SCT2, totSolarl, SolarW, PF, temp, freq, freeram

These fields do not match the defaults from the SD image or the instructions image for creating a device and setting up inputs and feeds.

I am using the 'solar' version, which looks like 2 identical sensor modules. I just have one hooked up to Main.

CircuitSetup, 2019-10-02:
This is being looked at by the EmonCMS people.

CircuitSetup, 2019-10-03:
This should be fixed with this PR: https://github.com/emoncms/device/pull/48

veggie2u, 2019-10-05:
There was an issue with the JSON that was fixed with https://github.com/emoncms/device/pull/49

closed 2019-10-05 by veggie2u