CircuitSetup/Split-Single-Phase-Energy-Meter #11: Can't Post Data from ESP to EmonCMS via MQTT

veggie2u, 2019-10-05:
I can't seem to get the data from the ESP to the mosquitto instance on my raspberry pi. I have in the EmonESP web page configured the server: emonpi (I can ping and ssh), the topic of emon/emonesp (I have a device set up in EmonCMS), and the user of emonpi, and password of emonpimqtt2016. I am pretty sure that the EmonCMS mosquitto is working. I found that from my Mac, this command will set the temp on the EmonCMS:

`mosquitto_pub -h emonpi.local -t emon/emonesp -m {\"temp\":\"20\"} -u emonpi -P emonpimqtt2016 -d`

I have also updated the firmware via the upload in EmonESP with the recent CircuitSetup solar bin file src_solar.ino.esp32.bin

Any ideas how to debug why it wont send data?

CircuitSetup, 2019-10-05:
There was a bug fixed early on with EmonESP early on, but im not sure why this wouldn't be working.  I assume you've seen this page? https://guide.openenergymonitor.org/technical/mqtt/

CircuitSetup, 2019-10-18:
Please reopen if this is still an issue.

closed 2019-10-18 by CircuitSetup

veggie2u, 2019-10-18:
Yes this is closed. I wanted to put the solution: you can’t use local extensions to host names from the ESP. The code posting to a url or MQTT is relying on an external server to do DNS. It does not use mDNS to check for a local url first. It is only used to broadcast the esp devices url. So you need to use an IP address, not a host name from the esp device. This is why the host name worked from a Mac, which checks for .local addresses, and not from the esp.