CircuitSetup/Split-Single-Phase-Energy-Meter #12: Can't complile EmonESP code using platformio

veggie2u, 2019-10-05:
Wanted to try logging to the serial monitor, so trying to compile the esp code using platformio. Is this what you are using @CircuitSetup ? Or the Arduino platform? As per the readme, I pulled the code, entered the EmonESP folder. If I try just a 
`pio run`
i am told that there is no emonesp environment, which is true. There is no definition for one in the platformio.ini file - there is in the upstream project. Since this is an Esp32 I changed the default to emonesp32 which does exist and tried to compile again with 
`pio run`
I then get compile errors
```compilation terminated.
*** [.pio/build/emonesp32/src/web_server.cpp.o] Error 1
src/energy_meter.cpp: In function 'void energy_meter_loop()':
src/energy_meter.cpp:175:29: warning: variable 'totalVoltage' set but not used [-Wunused-but-set-variable]
   float voltageA, voltageC, totalVoltage;
                             ^
In file included from src/http.cpp:32:0:
/Users/chris/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.h:29:1: error: expected class-name before '{' token
 {
 ^
/Users/chris/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.h:105:18: error: type 'Print' is not a base type for type 'WiFiClientSecure'
     using Print::write;
                  ^
src/http.cpp: In function 'String get_https(const char*, const char*, String, int)':
src/http.cpp:63:12: error: 'class WiFiClientSecure' has no member named 'print'
     client.print(String("GET ") + url + " HTTP/1.1\r\n" + "Host: " + host +
            ^
src/http.cpp:75:28: error: 'class WiFiClientSecure' has no member named 'readStringUntil'
       String line = client.readStringUntil('\r');
                            ^
*** [.pio/build/emonesp32/src/http.cpp.o] Error 1
```

Is this how you would/are building the code?

CircuitSetup, 2019-10-05:
I've only.compiled with Arduino lately because of some important updates to AayncTCP, that I dont believe are in platformio yet. This relate to memory leaks and not.closing browser sessions properly.

There should be a definition for the esp32 in the latest initiative file. I just added it recently. The total totalvoltage variable is only.used with 50hz setups, so that needs to be moved. The other errors, in not sure why they are coming up. Perhaps those libraries need to be updated.

veggie2u, 2019-10-05:
Ok. I have been trying to get this to compile under Arduino for a bit. I did run into all the issue from the Libraries Errors section of the readme. I also moved over the ATM90E32 files from the project into an Arduino library. 

I am stuck on the Hash library. The Hash.cpp you pointed to also has this line:
`#include <bearssl/bearssl_hash.h>`
Been trying to figure out where to get this. There is an ArduinoBearSSL in the library manager, but that doesn't look right. The Hash library in the ESP2866 board definition doesn't use bearssl, but it was trying to use <c_types.h> That doesn't work. I did find the file locally at
`~/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/tools/sdk/include/bearssl`
I have no idea how 'tools' are used, but the compiler doesn't seem to find it. Do you remember what you did? (Arduino 1.8.10)

CircuitSetup, 2019-10-06:
What version of the esp32 arduino IDE are you using? Hash should be included.  If not 1.0.3, use that as it has the latest fixes for a number of things.

veggie2u, 2019-10-06:
Do you mean the board? 1.04. I don't see it in 1.03 either though https://github.com/espressif/arduino-esp32/tree/1.0.4/cores/esp32

The Hash function you link to in the docs is from the esp8266 libraries
https://github.com/esp8266/Arduino/tree/master/libraries/Hash/src

It looks like it was changed recently from using an internal sha1 class (I am guessing you used that), to the bearssl based version. However, I can't get either one to compile. If I go to the branch tagged with 2.5.2, there is a separate file called sha1.c. This wont compile for me with the 
`#include <c_types.h>` line in there.
If I take that out, I then have issues with all the methods that have ICACHE_FLASH_ATTR in them. I am not a C programmer, and I have no idea what that is for. I do know that it is not defined in sha1.h. 

I have found it in the tools section of esp8266 there is a define there for ICACHE_FLASH_ATTR, but that c_types.h file is not in the section of the esp32 git library. I really can't see how this version of Hash can compile for you.

I tried adding the definition I found there:
```
#ifdef ICACHE_FLASH
#define ICACHE_FLASH_ATTR   __attribute__((section("\".irom0.text." __FILE__ "." __ICACHE_STRINGIZE(__LINE__) "." __ICACHE_STRINGIZE(__COUNTER__) "\"")))
#else
#define ICACHE_FLASH_ATTR
#endif /* ICACHE_FLASH */
```
No idea what that is doing though. It gets me past the compile, but the link fails. Not sure if that stuff is particular to the esp8266 or not.

I don't think this message is particularily useful, but this is what the linker shows
```
inking everything together...
/Users/chris/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc -nostdlib -L/Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib -L/Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority -u __cxa_guard_dummy -u __cxx_fatal_exception -Wl,--start-group /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/config.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/emoncms.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/energy_meter.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/http.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/input.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/mqtt.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/ota.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/src_solar.ino.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/web_server.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/sketch/wifi.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/EEPROM/EEPROM.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/ETH.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/WiFi.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/WiFiAP.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/WiFiClient.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/WiFiGeneric.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/WiFiMulti.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/WiFiSTA.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/WiFiScan.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/WiFiServer.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/WiFi/WiFiUdp.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/ESPmDNS/ESPmDNS.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/FS/FS.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/FS/vfs_api.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/Hash/Hash.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/Hash/sha1/sha1.c.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/SPIFFS/SPIFFS.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/AsyncTCP/AsyncTCP.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/ESPAsyncWebServer/AsyncEventSource.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/ESPAsyncWebServer/AsyncWebSocket.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/ESPAsyncWebServer/SPIFFSEditor.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/ESPAsyncWebServer/WebAuthentication.cpp.o /var/folders/gc/j9pm882j7852bw7gqpyg47yc0000gn/T/arduino_build_323319/libraries/ESPAsyncWebServer/WebHandlers.cpp.o / [... cut ...]
Using library PubSubClient at version 2.7 in folder: /Users/chris/Documents/Arduino/libraries/PubSubClient 
Using library ArduinoOTA at version 1.0 in folder: /Users/chris/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/ArduinoOTA 
exit status 1
Error compiling for board ESP32 Dev Module.
```
No idea why it says multiples of each library exists. If I remove the ones I put there, it is not found.

So, what board are you using? I have tried "ESP32 Dev Module" and "ESP32 Wrover Module"

CircuitSetup, 2019-10-08:
> Do you mean the board? 1.04. I don't see it in 1.03 either though https://github.com/espressif/arduino-esp32/tree/1.0.4/cores/esp32

Whoops, I meant 1.0.4, yes.

> The Hash function you link to in the docs is from the esp8266 libraries
> https://github.com/esp8266/Arduino/tree/master/libraries/Hash/src
> 
> It looks like it was changed recently from using an internal sha1 class (I am guessing you used that), to the bearssl based version. 

Ah, you're right, I didn't realize that was updated. I [added the version that I have in my libraries folder](https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/tree/master/Software/libraries/Hash) until I can look at this some more.

You shouldn't have to worry about the multiple library exists messages, as long as it's getting the correct one.

> So, what board are you using? I have tried "ESP32 Dev Module" and "ESP32 Wrover Module"

I have ESP32 Dev Module set.

veggie2u, 2019-10-09:
Found the errors. Hash is not needed on an ESP32. It is not included as a library in the core ESP32 code. Not copying it to your library, and commenting out the include in web_server.h will allow Arduino to build it.

To get it to build with PlatformIO, I believe I ran into a case insensitive problem. The project file called wifi.h imports WiFi.h. It looks like these were getting confused. When I rename the project file to wireless.h and wireless.cpp - and changed the correct references, the build works. As it appears to have built on Arduino, and not on PlatformIO, I am wondering if it is just PlatformIO or the combination of it on MacOS.

CircuitSetup, 2019-10-09:
ohh, okay - I'll make hash.h conditional to the esp8266.

I'll have to test if platformio gets confused with the wifi.h files on my PC. 

Thanks for figuring this out!

veggie2u, 2019-10-11:
Do you even want the esp8266 code? Your hardware is esp32. I could create a PR that removes the esp8266 ifdef stuff.

CircuitSetup, 2019-10-11:
There are others that use the esp8266. Thanks for offering.though!

tferrin, 2021-12-26:
@veggie2u With respect to "wifi.h" vs "WiFi.h", note that the macOS file system is case-insensitive. (It preserves the case in file names, but all name matching is case-insensitive.) Also, as far as I can tell, PlatformIO does not distinguish between #include <WiFi.h> and #include "WiFi.h". It always looks in the local src directory first. So the combination of these two things is what messed you up.

CircuitSetup, 2022-06-01:
This is fixed in the latest version of EmonESP

closed 2022-06-01 by CircuitSetup