CircuitSetup/Split-Single-Phase-Energy-Meter #14: Strange data shown by energy meter

gurari, 2019-10-11:
Hello,
I've just connected my energy meter (CT1 only used for the moment) and connected it to my home network. When I scrolled down to the latest data column I'm seeing strange data:

Key | Value
-- | --
totV | 1310.70
CT1 | 65.5350A
CT2 | 65.5350A
totI | 131.0700A
W | 0.0013W
PF | 0.00
temp | -1.00
freq | 655.35

I've tried resetting the meter and factory reset but still the same.. 
Any clues what I'm doing wrong?

gurari, 2019-10-11:
I assumed these values are some defaults when nothing is received from the software so I've opened the energy meter and found that nodemcu was not completely inserted into the socket. Maybe during the shipping something went wrong. Now all is ok when I pushed the nodemcu to the bottom.

closed 2019-10-11 by gurari

CircuitSetup, 2019-10-11:
I was going to suggest making sure it was seated. Glad you figured it out!