CircuitSetup/Split-Single-Phase-Energy-Meter #17: External wifi antenna?

BitKnitting, 2019-10-16:
The ESP32 freezes up after about 7 hours (or so...) I'm thinking it is because the wifi connection is not strong enough.  Has anyone put an external antenna on their ESP32?  If so, any recommendations, gotchas?

Thank you.

CircuitSetup, 2019-10-16:
What specific software are you running?  Also what is the rssi value?

There is an esp32 dev module with an external antenna connection.

gigawatts, 2019-11-28:
Speaking of RSSI, it would be awesome to add that as one of the values available via MQTT, so it can be graphed/monitored over time.

CircuitSetup, 2019-12-23:
@BitKnitting I meant to post this version of the DevKitC- https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-32U?qs=sGAEpiMZZMuJ3l9lTgMBp1ZUJQFPkBG2sDWyUw9oyIyrWi1BCP7QAQ%3D%3D

closed 2020-03-09 by CircuitSetup