CircuitSetup/Split-Single-Phase-Energy-Meter #19: Power factor value alternating +ve/-ve in ATM90E32.cpp

gbeardall, 2019-11-10:
The power factor value alternates between a +ve & -ve value for a constant load.
There is a bug in the precedence of the  `if()`  statement in all power factor functions.
The '&' operation needs to be surrounded by brackets, see below:

```
double ATM90E32::GetTotalPowerFactor() {
  signed short pf = (signed short) CommEnergyIC(READ, PFmeanT, 0xFFFF);
  //if negative
  if ((pf & 0x8000) != 0) {
	pf = (~pf) + 1;
  }
  return (double)pf / 1000;
}
```

Also, why is there effectively an `abs()` operation being performed on these values, doesn't a -ve value represent an inductive load?

Thanks, Gavin

CircuitSetup, 2019-11-10:
Thanks for bringing this up @gbeardall - I was literally just looking into this because I realized it didn't make sense. I _think_ this was done because the original function example (in C) from Atmel looks like this:

```
          val = Read_AFE_1Reg((uint16_t)P3_PFmeanT);
          if ( (val & 0x8000) == 0x8000 )
          {
            val = (~val) + 1;
            error_type = 2;
          }
          else
          {
            error_type = 1;
          }
          val = (val & 0x7FFF);
          //ptr =&VAFE.RMS[PFt];
          Calculate_PowerFactor( ptr, val );
```

Then Calculate_PowerFactor is:

```
void  Calculate_PowerFactor( uint32_t *rmsptr, uint32_t rmsval )
{
  uint32_t  m;

  m = (rmsval & 0x7FFF);
  *rmsptr = (m % 10000) / 1000;
  *rmsptr = *rmsptr << 4;
  *rmsptr |= (m % 1000) / 100;
  *rmsptr = *rmsptr << 4;
  *rmsptr |= (m % 100) / 10;
  *rmsptr = *rmsptr << 4;
  *rmsptr |= (m % 10);
}
```

And they do something similar to get most metering values. I'm puzzled as to why, since you're absolutely correct. The values can be negative when there is an inductive load. I previously corrected the other power values.

Either way, I'll remove the bitwise lines in the library.

gbeardall, 2019-11-10:
That's some weird looking code!
As far as I can see it's converting the power factor to BCD in rmsptr

Do you have a link to the original Atmel example code you referenced, so I can take a look?

gbeardall, 2019-11-10:
Just tried it now with an inductive & a capacitive load and the sign of the power factor now changes appropriately.

CircuitSetup, 2019-11-10:
> That's some weird looking code!
> As far as I can see it's converting the power factor to BCD in rmsptr
> 
> Do you have a link to the original Atmel example code you referenced, so I can take a look?

Yes, it is. They do some things in a very roundabout way, and theres a lot of repetative code. Maybe they wanted to do bit shifting for some reason, and that's why it all had to be positive? I'm not sure. This is a 16 bit register, but there are ones that are 32, where bits do need to be shifted because the values are stored in 2 different registers.

It's not available anywhere (Microchip had it behind a login that I had to request access to) so I'll have to post it somewhere private.

CircuitSetup, 2019-11-10:
> Just tried it now with an inductive & a capacitive load and the sign of the power factor now changes appropriately.
> 
> Although, it is a bit confusing as the sign does also change depending on which direction you have the current clamp.

I did the same and it looks correct.

CircuitSetup, 2019-12-03:
Corrected in https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/releases/tag/v2.5.5

closed 2019-12-03 by CircuitSetup