CircuitSetup/Split-Single-Phase-Energy-Meter #20: 240V and 60Hz?

CircuitSetup, 2019-12-03:
But what about 240V and 60Hz?

_Originally posted by @netomx in https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/issues/7#issuecomment-537653582_

chaseadam, 2020-10-02:
I can confirm this board and software works for *split* phase 240V at 60hz in North America (most people think there are 2 phases when it is split phase, but true 2 phase is quite rare as if you have more than 1 phase, you generally have 3 phases 120° offset). Just make sure you set the calibration values as recommended for 60hz and the chosen transformer.

netomx, 2020-12-28:
> I can confirm this board and software works for _split_ phase 240V at 60hz in North America (most people think there are 2 phases when it is split phase, but true 2 phase is quite rare as if you [...]

What transformer did you use? I havent found anything with 240V

chaseadam, 2020-12-28:
I stuck with measuring just one of the two legs for the voltage (so a 120v transformer). 

Ideally each leg is measured individually. Measuring between the legs (240v) is no better than measuring one leg as current flows through the neutral in a split system.

I do not have the reference, but I recall reading that there is negligible gain by measuring individual legs. I can imagine if you have a severely unbalanced panel it may make a difference.

missmah, 2021-12-04:
I've seen 125V on one phase and 108V on the other phase (at the same time) on my panel in real scenarios. Also, when combined, I've commonly seen 205-218V, so likely 120* offset (the building is 3-phase, each unit gets 2 of the three phases). In any case, I think metering one phase and multiplying by a constant would be very inaccurate in my situation.