CircuitSetup/Split-Single-Phase-Energy-Meter #21: Password Reset

catalinalb, 2019-12-07:
Hi

How can I reset the login password and admin user? I did set them up but the forget about them. 
I think the only way would be to update the ESP from the Arduino IDE can someone let some instructions about this procedure?

Thank you
Catalin

CircuitSetup, 2019-12-08:
To do this you'll have to reset things to default. This will erase any saved settings, like wifi configuration and connections.

1. Remove the ESP32 off of the pcb
2. Power the ESP32 via USB (If you want, open a serial window to see what it's outputting)
3. Connect Gnd (upper left) with GPIO3 (5th pin from upper left) with a jumper wire
4. Wait 15 seconds (10 minimum, but 15 to be sure). If you have a serial window open, it'll say "factory reset"

closed 2019-12-08 by CircuitSetup

catalinalb, 2019-12-08:
It Worked!! 

No need to desolder the ESP since it´s connected with pin headers (plug&play) :)
Baudrate 115200
Here is a link with the pinout: (GPIO3 also labeled RX)
https://www.flickr.com/photos/jgustavoam/40089095211

Thank you,
Catalin

CircuitSetup, 2019-12-08:
Great! Glad it worked for you