CircuitSetup/Split-Single-Phase-Energy-Meter #22: Feature Request: RSSI value MQTT feed

gigawatts, 2019-12-10:
I originally asked about this in #17 but it probably makes more sense to track in its own thread.

It would be awesome to add the WiFi RSSI as one of the values available via MQTT, so it can be graphed/monitored over time.

CircuitSetup, 2019-12-23:
That is a good idea! It shouldn't be that hard to add this to the feed.

fredkehler, 2020-01-16:
I have added rssi to my mqtt, if you replace your mqtt.cpp with this file and recompile you should be good, 
Note:  I was having trouble losing wifi connection, and took out the blocking loop on mqtt reconnect which solved my problem... if you need it, the code is still there, just remove the rems....

[mqtt.zip](https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/files/4072035/mqtt.zip)
cheers

gigawatts, 2020-01-16:
I assume that will make it into an upcoming binary release as well? My WiFi connection to the ESP is just on the edge of "usable", where it will work for several days, then fall off the network and fail to ever re-join, requiring a power-cycle to bring it back online. Is that similar to the trouble you are experiencing? 

My plan was to setup another AP closer to the ESP so it would stop falling off the network, but it sounds like you may have corrected the issue in code?

fredkehler, 2020-01-16:
your experience sounds very similar to mine... I moved my esp to within 5ft
of my AP and still couldnt keep a connection going for even 24 hrs...
my mods at this point have been working for 8 days without a hickup... so I
am cautiously optimistic... I would gladly share my code... but I am not
familiar with using github and how the versioning system works... I have
kept mine versioned as 2.5.5fk just to differentiate it from the main
code...
are you setup to compile on arduino or would you just like to try the
binary?

gigawatts, 2020-01-17:
> your experience sounds very similar to mine... I moved my esp to within 5ft of my AP and still couldnt keep a connection going for even 24 hrs... my mods at this point have been working for 8 days without a hickup... so I am cautiously optimistic... I would gladly share my code... but I am not familiar with using github and how the versioning system works... I have kept mine versioned as 2.5.5fk just to differentiate it from the main code... are you setup to compile on arduino or would you just like to try the binary?
> […](#)
> -- Alfred (Fred) Kehler ( fred.kehler@gmail.com)

I responded to your comment before realizing you were not the original code author 🤣 That's why I originally mentioned the binary release, then removed it once I looked at your name.

I do have PIO setup to compile for ESP chips, I'll give your mods a shot soon. My ESP has been stable for the last several days after setting up a closer AP, but I would rather not have to run a second AP if I don't have to.

The easiest way to contribute back to the project is just to hit the fork button at the main repo, edit whatever you like in your own copy, then it's pretty easy to "compare" back to this repo and submit a pull request. It can all be done from the web UI if you are not comfortable with git commands yet. If the author accepts your changes, they can tweak version numbers before merging.

fredkehler, 2020-01-17:
appreciate your comments... what would you suggest as the best way to fix
this?  my github account does not appear to allow me to use the 'fork'
option ( says I am not a member of an organization?) and I don't see any
options to allow me to delete or retract what I have submitted...?

gigawatts, 2020-01-17:
It looks like you figured it out yesterday? Your fork is here: https://github.com/fredkehler/Split-Single-Phase-Energy-Meter

CircuitSetup, 2020-02-21:
Thanks @fredkehler - this is in the latest version. 

I also fixed the issue with the reconnect blocking everything. There were extra 0s on the counters. Instead of 5 seconds, it was trying to reconnect continuously for 14 min, then would have tried again every 28 min. This was likely causing things to lock up, and/or just blocking everything from running.

closed 2020-02-21 by CircuitSetup

gigawatts, 2020-02-22:
Is there anything special we need to do with the spiffs file in the latest release? Do we need to upload both src.ino.esp32.bin and src.spiffs.bin, but in a certain order?

CircuitSetup, 2020-02-24:
> Is there anything special we need to do with the spiffs file in the latest release? Do we need to upload both src.ino.esp32.bin and src.spiffs.bin, but in a certain order?

The order shouldn't matter. The spiffs update just changes some text for default calibration values, so it's not really necessary either.