CircuitSetup/Split-Single-Phase-Energy-Meter #23: FR:  Adjust Publish Rate for MQTT

politekcmo, 2019-12-20:
It would be fantastic if there was a way to adjust how frequently the ESP publishes to MQTT from the device itself.  It's quite fast right now, and that is definitely desirable but I would love to adjust it down for applications that have less bandwidth available.

politekcmo, 2020-01-18:
In my testing this works right, but I believe this is a solid way to allow for a customizable delay.  Would you mind looking it over?  https://github.com/politekcmo/Split-Single-Phase-Energy-Meter

CircuitSetup, 2020-05-27:
Adding an actual delay would block other functions from running. A better way to do it would be to compare a publish delay (or publish every x seconds) against millis.

Also, the MQTT library has been changed in the latest version to add one that uses async (so it doesn't block other things from running).

nightshade00013, 2022-09-24:
If you are using ESPhome it's simple to include this near the top of the YAML in the substitutions section

`update_time: 10s `

Then add the following for each sensor under each "- platform:" section
`update_interval: ${update_time}`

Then all you have to do is change the reporting time to how often you want it to happen.  Shouldn't matter if you are using MQTT or the HomeAssistant/ESPhome API's  You can also hard set the "update_interval:" on each section individually and disregard the "update_time" setting altogether.

https://github.com/nightshade00013/ESPHome_EnergyMonitor/blob/master/NightShade4chan32.yaml