CircuitSetup/Split-Single-Phase-Energy-Meter #28: No Hardware

palmtreefrb, 2020-02-20:
# Order 89604 Invoice
# Date Placed on Nov 10, 2019
What is happening to deliveries from Crowd Supply? Ordered in Nov 2019, every week delivery date keeps getting pushed back.

### Originally expected to ship on Dec 15, 2019
### Currently expected to ship on Feb 21, 2020

CircuitSetup, 2020-02-21:
I'm sorry it's taking so long to get your order shipped. I was told there were some delays on the logistics side of things, but that orders were expected to ship very soon.

bateschris, 2020-02-27:
I'm in the same boat, only my order has changed overnight last night from Feb 21, 2020 to MARCH 20, 2020. Crowd Supply has twice replied to me saying that they have parts from you, and they were sent to their warehouse ("last week"), so I'm pretty confused as to what the delay is.

CircuitSetup, 2020-02-27:
I'm not really sure what the delay is, but the good news is things were [listed on Mouser.com](https://www.mouser.com/new/crowd-supply/crowd-supply-circuitsetup-energy-meter/), who does the fulfillment for Crowd Supply.

McGeaverBeaver, 2020-03-15:
I ordered Oct 27 2019 and I got my device 2 weeks ago.. Shipped Feb 28 2020.  👍  and it's pretty sweet.

palmtreefrb, 2020-03-15:
Finally received it as well....

closed 2020-03-15 by palmtreefrb