CircuitSetup/Split-Single-Phase-Energy-Meter #30: Voltage * Current != Active Power, is that correct?

manutenfruits, 2020-03-10:
I am using the [Expandable 6 Channel ESP32 Energy Meter](https://circuitsetup.us/index.php/product/expandable-6-channel-esp32-energy-meter/) with 2x [120A/40mA Current Transformer YHDC SCT-016](https://circuitsetup.us/index.php/product/120a-40ma-current-transformer-yhdc-sct-016-with-3-5mm-jack-16mm-opening/) and **ESPHome**.

The problem I have is that while Voltage and Current match that of my golden Kill-A-Watt, the Wattage doesn't equal Voltage * Current.

Reported by the sensor:
Voltage Phase A: 123.37 V
Current A: 0.53 A
Active Power A: 8.27 W

Calculated
123.37V * 0.53A = **65.38W != 8.27 W**

Am I understanding this incorrectly? I know I'm using ESPHome but I want to understand I got the theory right first.

CircuitSetup, 2020-03-10:
What do you have for your config file in ESPHome?  Also, what version of the meter? 

It looks like one of the CT's is backwards. Normally one of the currents would report as negative if this were the case, but if you have a multiplier in the config, it'll come out as positive. It could also be on the opposite 

There is another variable that the meter takes into account when calculating active power. Power factor. (Voltage * Current) * Power Factor = Active Power

A perfect power factor, that of a pure resistive load, is 1.

manutenfruits, 2020-03-10:
So indeed I had both clamps with the arrow pointing towards the distribution box with a `multiply: -1` filter, I fixed it and removed the filter but I can't tell if anything changed. I added a Voltage on phase_b with the gain just to have a second measurement (reports similar voltage).

Voltage A | 123.27 V 
Current A | 0.78 A 
Active Power A | 13.82 W 
Voltage B | 123.32 V 
Current B | 1.02 A
Active Power B | 102.81 W

How is power factor calculated? is it a parameter or is it something that the chip can measure?

Here is my config, and I'm on the `dev` channel

```

sensor:
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: "Voltage A"
        id: voltage_a
      current:
        name: "Current A"
        id: current_a
      power:
        name: "Active Power A"
        id: active_power_phase_a
      gain_voltage: 7314
      gain_ct: 21111
    phase_b:
      voltage:
        name: "Voltage B"
        id: voltage_b
      current:
        name: "Current B"
        id: current_b
      power:
        name: "Active Power B"
        id: active_power_phase_b
      gain_voltage: 7314
      gain_ct: 21111
    frequency:
      name: "Line 1 Frequency"
    line_frequency: 60Hz
    gain_pga: 2X
    chip_temperature:
      name: "Temperature"
    update_interval: 5s
  - platform: template
    name: "Calculated Power A"
    id: calculated_power_a
    unit_of_measurement: W
    update_interval: 5s
    lambda: |-
      return id(voltage_a).state * id(current_a).state;
  - platform: template
    name: "Calculated Power B"
    id: calculated_power_b
    unit_of_measurement: W
    update_interval: 5s
    lambda: |-
      return id(voltage_b).state * id(current_b).state;
```

manutenfruits, 2020-03-10:
Okay I got it all figured out, there are some gaps on the esphome docs but I added the `reactive_power` and `power_factor` measurements and now it should add up. Thanks for your help!

CircuitSetup, 2020-03-11:
Great! Glad you figured it out! 

FYI, Power factor is calculated by the meter. [Here's the math of how it is done.](https://www.allaboutcircuits.com/textbook/alternating-current/chpt-11/calculating-power-factor/)
(true power = active power) So you can see why if current is negative, that would throw everything off.

closed 2020-03-11 by CircuitSetup