CircuitSetup/Split-Single-Phase-Energy-Meter #31: emonESP split phase - can I use for single phase?

m-d-johnson, 2020-03-10:
Hi
I've got my hands on one of these, but I have a simple use case in which I just need to watch the current going into the distribution board. I have two CT16 current sensors and an idealpower DB-06-09 AC/AC adapter. 

I've never set one of these up before. Is this the right place to be asking for help?  I've tweaked the calibration figures and the thing seems to be reading some unusually high values.  Power is off by an order of magnitude at some points.

CircuitSetup, 2020-03-11:
Okay, first, some questions: 

1. Do you have the kit meter? 
2. If so, in the EmonESP web interface, do you have the setting set for 50Hz? 
3. Specifically, what values are high, and what are they? 
4. What version of the meter do you have? (it's probably v1.4 rev2 or rev3 - rev3 will have lower voltage calibration values)

m-d-johnson, 2020-03-11:
Thanks for the response!  I've got the pre-assembled kit, which came with the CT clamps but not the AC adapter as I'm in the UK and I have a 240V 50Hz supply.  

I see the setting for 50Hz and I've set it accordingly.

Watts is high, and current is correspondingly high too.  At one point my smart meter was reading about 500W and the emonESP was reading 5kW or so.  

Do I use one clamp or two?

The version - can I find it in software?

CircuitSetup, 2020-03-11:
If you have a single phase, you should only use 1 clamp around your main wire. 

The version of the board is printed on the pcb itself, like here: https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/raw/master/images/energy_meter_small.jpg

Or if you tell me when (and where) you purchased the kit I can probably figure out what version it is.

m-d-johnson, 2020-03-11:
Great. I'll check the clamp/board shortly.  I bought it in December (it was on a pre-order) from Crowdsupply.

m-d-johnson, 2020-03-11:
1.4 rev 3

m-d-johnson, 2020-03-11:
OK this is a little embarrassing, but it seems to be reading accurately (give or take a few watts) now. Maybe I hadn't cycled it enough times or something. I've got 389 in the frequency field, and the rest at their defaults.

closed 2020-03-11 by m-d-johnson