CircuitSetup/Split-Single-Phase-Energy-Meter #34: No readings from CT2

neilrees, 2020-04-11:
Hi,

I'm attempting to install the sensor in the UK on a 240v single phase supply.  The intent was to monitor the mains supply with CT1 and the PV Generation with CT2  (and use this to calculate amount being imported, exported, and consumed)

The meter is connected up and I'm getting good readings from CT1 but I just get no readings from CT2.

I've flashed ESPHome firmware onto it using this yaml:

```
esphome:
  name: [name]
  platform: ESP32
  board: nodemcu-32s

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
  - platform: atm90e32
    cs_pin: 5
    phase_a:
      voltage:
        name: "Line Voltage A"
      current:
        name: "CT1 Current"
      power:
        name: "CT1 Active Power"
      gain_voltage: 7243
      gain_ct: 25498
    phase_b:
      voltage:
        name: "Line Voltage B"
      current:
        name: "CT2 Current"
      power:
        name: "CT2 Active Power"
      gain_voltage: 7243
      gain_ct: 25498
    frequency:
      name: "EMON Line Frequency"
    line_frequency: 50Hz
    gain_pga: 2X
    update_interval: 2s
```

If I switch the two CT clamps over the problem remains on CT2.

I get the following log output from ESPHome:

```
[14:17:24][D][sensor:092]: 'Line Voltage A': Sending state 246.80000 V with 2 decimals of accuracy
[14:17:24][D][sensor:092]: 'Line Voltage B': Sending state 0.01000 V with 2 decimals of accuracy
[14:17:24][D][sensor:092]: 'CT1 Current': Sending state 4.06200 A with 2 decimals of accuracy
[14:17:24][D][sensor:092]: 'CT2 Current': Sending state 0.00100 A with 2 decimals of accuracy
[14:17:24][D][sensor:092]: 'CT1 Active Power': Sending state 934.28766 W with 2 decimals of accuracy
[14:17:24][D][sensor:092]: 'CT2 Active Power': Sending state 0.00000 W with 2 decimals of accuracy
[14:17:24][D][sensor:092]: 'EMON Line Frequency': Sending state 49.96000 hz with 1 decimals of accuracy
```

Any ideas on what I've missed in order to get CT2 working?

CircuitSetup, 2020-04-11:
Change phase_b to phase_c.

neilrees, 2020-04-11:
That's it!  Thankyou!

I was working from the example on ESPHome's site: 
https://esphome.io/components/sensor/atm90e32.html

But to be fair, the example linked to from this project is correct:
The guidance linked here is correct https://github.com/digiblur/digiNRG_ESPHome/blob/master/digi_nrg_2chan32.yaml

Thanks!

CircuitSetup, 2020-04-11:
Yeah, the first version of the docs were wrong since this meter doesnt use phase_b. I corrected it in https://next.esphome.io/components/sensor/atm90e32.html

closed 2020-04-11 by neilrees