CircuitSetup/Split-Single-Phase-Energy-Meter #36: Can we current with one CT only?

aaamitsingh, 2020-08-17:
In the description you have mentioned that two CT in opposite direction is required to measure the current.  Is this possible to do the same with the help of only one CT?

CircuitSetup, 2020-08-17:
Do you mean 2 wires in 1 CT? If so, yes, this can be done as long as the CT is large enough and the 2 wires are fed through it in opposite directions. If they are the same direction power factor will be off and power will be calculated incorrectly.

aaamitsingh, 2020-08-17:
Thanks for your response.  No I mean that I want to measure current with the one CT only. I don't want to use the second CT in opposite direction. 
Is there any way to read the pf and power correctly with one CT only?

CircuitSetup, 2020-08-17:
Do you have split single phase power or single phase?

aaamitsingh, 2020-08-17:
Single phase

CircuitSetup, 2020-08-17:
You shouldn't have a problem only using 1 CT then. The meter will output power factor and power for that channel only. 

The 1 opposite CT only applies to split single phase power.

aaamitsingh, 2020-08-17:
Great that means I can also monitor the current consumption of 2 different device with single board using 2 different CT?

http://www.hlktech.net/product_detail.php?ProId=64 
Can we use the above ultra small capsule transformer instead of mentioned bulky transformers?

CircuitSetup, 2020-08-17:
You can as long as you are not reading the total power. There are outputs for single power channels in the Arduino library, and with ESPHome. EmonESP, by default, outputs the total wattage. 

No, that power module will not work. It has to be an AC to AC transformer with at least 1000mA, and between 9-12VAC output.

aaamitsingh, 2020-08-18:
Thank you so much for clearing all my queries and fast response.  I'm closing this issue now but once we develop this product I will share the feedback with you.

Thank you one again.

closed 2020-08-18 by aaamitsingh