CircuitSetup/Split-Single-Phase-Energy-Meter #37: Can't flash latest esphome example

jarrettv, 2020-09-30:
I'm using the latest example code here: [Software/examples/HA-ESPHome_energy_meter_detailed_power.yaml](https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/blob/e1790d1bc7859581c94a0c7407c2ae6370ee8dd2/Software/examples/HA-ESPHome_energy_meter_detailed_power.yaml) and I get an error during upload. I just tested with another esp8266 and it flashed fine.

`CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading .pioenvs/energy_meter/firmware.bin
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
============================================== [FAILED] Took 26.92 seconds ==============================================`

jarrettv, 2020-09-30:
I was able to grab the firmware.bin and flash it from emonesp.local flash utility and it worked!

closed 2020-09-30 by jarrettv

CircuitSetup, 2020-09-30:
Did you mean to load EmonESP or ESPHome? Theyre both very different.

jarrettv, 2020-10-26:
I was running emon and I wanted to switch to esphome. I was able to use emon firmware upgrade to flash esphome. Hope that explains it.