CircuitSetup/Split-Single-Phase-Energy-Meter #38: EmonESP fork

chaseadam, 2020-10-02:
It appears there are specific changes to a snapshot copy of upsteam EmonESP. I presume this is a snapshot (from 17 moths ago?) of this fork https://github.com/CircuitSetup/EmonESP?

It appears there is active work on https://github.com/CircuitSetup/EmonESP/commits/v3.0_merge presumably to move from the current EmonESP v2.6 to v3.x. Can you confirm this is the location being used to merge from upstream?

Thanks for the complete solution (hardware, software and docs!)

CircuitSetup, 2020-10-02:
Glad you noticed! Yes, I'm currently merging some changes for the most recent version of EmonESP. The calibration fields still have to be added to the gui, which is completely new.

closed 2021-01-11 by CircuitSetup

chaseadam, 2021-08-11:
If I want the latest and greatest, should I build and flash "master" on EmonESP? I see it was mentioned in #27