CircuitSetup/Split-Single-Phase-Energy-Meter #40: Verifying Power Factor values

chaseadam, 2020-10-06:
I believe with the Microchip IC used, all ADC values are captured simultaneously. This would imply that power factor calibration is largely unnecessary unless the AC-AC transformer is introducing some phase delay. I have a very low power factor seemingly correlated to the output of my PV microinverters. It even is close to zero for nearly 30 minutes.

Has anyone else reported similar experience with solar inverters?

What has been your experience with phase shift settings? Suggestions for calibration/validation?

[image]

CircuitSetup, 2020-10-06:
Some questions...

1. Are you measuring the use from your PV system?
2. Do you have a split single phase electrical system?  
3. If so, are you measuring both phases?

It makes sense that PF for mains would go negative if you're exporting power to the grid. When it's at 0 you aren't importing or exporting power - only the PV power is being used. PF slowly increases as you use less PV power, and more grid power.

chaseadam, 2020-10-06:
Seems like everything is working as intended then.

I am not measuring the PV system yet (I have the second board, I just haven't installed it yet)

Yes, I have a split single phase 240v 60hz (North American houshold) with 200A service.

Yes, I am measuring both legs of the split phase.

I understand that the PF would go negative on "export" of PV power, but I don't follow why the PF gradually transitions through 0 as the PV generation goes down. Sounds like I need to do more research.

closed 2020-10-06 by chaseadam

chaseadam, 2020-10-08:
A bit of research showcases why the Power Factor drifts through 0 as PV generation goes up. The key is the PF is the *relationship* between real and apparent power. Changing the "real" power value with a device with a PF near 1 (i.e. water heater) or -1 (PV inverter) can significantly impact PF without any affect on apparent power.

If apparent power stays the same (is not corrected) and real power is reduced (either by PV generation with a PF of 1 or by unplugging a device which is consuming real power with a PF of 1), the *ratio* of real to apparent power changes, reducing power factor. The triangle diagram vs the "phase" presentation helped me understand it.

[image]
[image]

https://www.sma-sunny.com/en/bad-power-factor-a-reason-to-oversize-your-inverter/

While there are PV inverters which can also actively correct PF, I do not have a model which supports that function.

CircuitSetup, 2020-10-08:
Thanks for posting that! Very good explanation.