CircuitSetup/Split-Single-Phase-Energy-Meter #42: Can't connect to Emoncms after updating from 2.5.7 to 2.6

zirdum, 2020-11-09:
I updated via the web interface (from 2.5.7 to 2.6, using the v2.6-src.ino.esp32.bin file) and it seemed to update just fine. After coming back up, I notice the MQTT section says it's connected as it was before the update, but the Emoncms says it's not connected. I retyped all the settings (server, node name and api key) to make sure it's proper, but still no go. Any ideas what changed that it can't connect at this point?

CircuitSetup, 2020-11-09:
I'm sorry the meter is giving you issues. 

Nothing has changed between versions in the EmonCMS connection code.  Can you double check the API key, username/password?

cap9qd, 2020-11-09:
To share my experice with the upgrade to 2.6.

The esp software update all went well and I didn't have to update my
SPIFFs. However, I did have to reenter my api key and readd the device in
my RPi EmonCMS install. However, readding the device may have been due to
updating EmonCMS to the latest and swapping to the "metering" version of
the ESP software.

zirdum, 2020-11-10:
I did double check all the fields, and it was entered in properly. I also power cycled it to make sure.

I downgraded back to 2.5.7 and it connected automatically after it restarted.

CircuitSetup, 2020-11-13:
I'm honestly not sure why that happened. It's possible the memory was corrupted when moving to the new version.

closed 2021-02-04 by CircuitSetup

dimaj, 2021-08-12:
sorry to bring up old thread, but i'm having same results...
I finally got around installing my Split Single Phase Energy Meter with Solar device. Today, I've updated to 2.6 and lost connectivity to my emoncms. Complete system reset didn't help. Downgrading to 2.5.7 worked.

zirdum, 2021-08-13:
I haven't tried updating it since I had the issue. I wonder if there could be some kind of handshake/auth value that needs to be reset (I'm guessing on the emoncms end)?

dimaj, 2021-08-13:
I looked at what code has changed, and I saw that http stack has been overhauled. I host emoncms locally and I have configured emonEsp to access emoncms via ip address.

how would one enable and access debug logs?