CircuitSetup/Split-Single-Phase-Energy-Meter #43: What Calibration setting value to use for 200A CT?

JPElectron, 2020-11-13:
Per 7. Calibration Settings...

25498 - SCT-013-000 100A/50mA
39473 - SCT-016 120A/40mA (default)
46539 - Magnalab 100A

...so what do I put if the CT sensor is this 200A version: https://circuitsetup.us/index.php/product/200a-100ma-current-transformer-yhdc-sct-024-24mm/

Also, if CT1 reports - (negative) value and CT2 is positive, does one need to be flipped?  Documentation is contradictory...
Picture shows arrows pointing opposite ways on each leg of split phase, but troubleshooting says...

I'm getting a very low power factor reading, and wattage seems off

One of your CT clamps is probably backwards - flip it around. **_If they are both oriented in the same direction, this will not happen._**

CircuitSetup, 2020-11-13:
Thanks for pointing out the fact that this value is missing. I'll have to add it to the documentation, and update things to not contradict. 
It should be around 26315

The CT that is reading negative should be turned around.

fjmohammed, 2021-01-03:
I am planning on using the CT referred here (https://circuitsetup.us/index.php/product/200a-100ma-current-transformer-yhdc-sct-024-24mm/). Since they have the TVS diode, I do not need to sever the burden resistor, correct?

CircuitSetup, 2021-01-11:
> I am planning on using the CT referred here (https://circuitsetup.us/index.php/product/200a-100ma-current-transformer-yhdc-sct-024-24mm/). Since they have the TVS diode, I do not need to sever the burden resistor, correct?

No, you don't need to (and shouldn't) sever the burden resistor.

closed 2021-01-11 by CircuitSetup

giulianomotter-trucks, 2021-02-03:
Hello,

> It should be around 26315

Could you give some insight about how you come up with these values?

The calibration process suggested in the AppNote needs an input of the value read on reference meter, and also value read on the ATM90 chip, so I'm kind of curious how can you pre-define these values?

Thank you.

JPElectron, 2021-02-03:
The 26315 is a starting point...

You need to calibrate it by putting a clamp amp meter around the mains cable, then manually adjusting the 26315 number +/- until the amps measured by your meter and the emonESP match - at least close.  You can see the live value as read by the emonESP by hitting it's IP in a browser directly.  It's sorta a pane in the butt cause obviously your power use is fluctuating in the house as you do it.

In my case, using the emonESP, with the 200A CTs it shipped with, my values turned out to be as follows...

Voltage: 3920
CT1: 26100  (I found 26315 to be off a bit)
CT2: 26100
Freq: 4485
CT Gain (PGA): 21  (I found 42, as the webpage suggests to be way off)

Some math is being done against the 26315 number to calculate the amperage displayed.  That's why adjusting this number +/- helps make it more accurate to actual amps being used.  There are likely differences in the way the CTs are manufactured, but also how you attach them (snug to the mains cable), and electrical noise that makes the case for calibration against another known source (amperage clamp style meter) necessary.

Side note - at some point I flashed mine with the latest firmware file (more difficult to find than it needed to be) and ended up "downgrading" mine to v2.6 whereas it shipped with v2.6.1  :(