CircuitSetup/Split-Single-Phase-Energy-Meter #44: Power is showing negative now?

scfigg, 2020-12-28:
I've had this running just fine for a while now - and a couple weeks ago, it started showing power in the negatives. Not sure what's going on here. I've of course tried power cycling everything - any thoughts?

[](url)
[image]

rstanley75, 2021-01-11:
I just updated to 2.6 from 2.5.4 and suddenly I started having the same problem with negative power values.

[image]

CircuitSetup, 2021-01-11:
1 of two things will show negative power: 

- You're exporting power to the grid (usually only happens with solar grids)
- A CT (or both) are backwards

@rstanley75 Previously in v2.5 there was a correction for any negatives. This was removed so that the power reading would calculate correctly.

scfigg, 2021-01-11:
Can I ask you to clarify a CT being backwards? Do you mean plugged into the wrong 3.5mm port or configured incorrectly? I had it working correctly for me for the past year or so before any upgrades.

CircuitSetup, 2021-01-11:
@scfigg literally the CT is backwards on the wire and has to be flipped around. If you were to plug the AC transformer into a plug on the opposite phase, this would also correct the negative issue.

rstanley75, 2021-01-11:
> @scfigg literally the CT is backwards on the wire and has to be flipped around. If you were to plug the AC transformer into a plug on the opposite phase, this would also correct the negative issue.

Now that I'm thinking about it, when I was rewiring some circuits during a kitchen remodel I moved the breaker for the outlet my AC transformer is connected to.  At the time I hadn't given any thought to the AC transformer but I know it's on the other half of the split-phase from where it was originally.   I bet the old firmware was covering up my mistake.

scfigg, 2021-01-14:
I guess I'm not fully understanding what is backwards. I rotated both of them around so that the it's 180 degrees rotated from where it was - and that didn't work. Still negative values. Why am I stupid? What am I doing/reading wrong here?

https://imgur.com/a/dPQh6eZ

scfigg, 2021-01-14:
Nevermind - it was upside down. I guess that was the last thing to try! Looks like I'm getting data correctly now!

closed 2021-01-14 by scfigg