CircuitSetup/Split-Single-Phase-Energy-Meter #45: Reading 65k from voltage register

TSMotter, 2021-01-30:
Hello, I'm trying to read significant data from my mains power system using a custom ATM90E36 based board I've designed.

I'm using your repo as a reference, even though the chip used here is ATM90E32, and mine is 36. 

On your main readme file you mention about readings on voltage register being something like 65k and mention that this normally indicates that something is not hooked up or working correctly which sugests to me a hardware problem. 

I'm currently getting this kind of reading. Was wondering if you have some troubleshooting tips for me, or if you could even help me with a simple hardware review of my board/setup.

Thanks.

CircuitSetup, 2021-01-30:
If you're getting all 65535 as output, it's either not communicating via SPI, or there is a pin on the IC that is not grounded properly. 

Make sure your MCU is using the correct SPI mode, and that the pins are all pulled high.

For the 36, you have to ground the DMA pin (36) if you're not using it. I made the mistake of not doing this when using that IC instead of the 32 for another project.

TSMotter, 2021-01-30:
Thak you for your reply.

Unfortunately that is not the case. I can communicate with the IC. I see all communication being made through a Logic Analyzer. So it's not SPI related. I can read/write to all registers with no problem.

The readings I'm getting on voltage are something like 65505, not all the way to 65535.

I've attatched a sketch of my schematic, if you would be kind enough to take a look and see if you have any observation I'd be very grateful.

Some information: 
- Mains is 3 phase 127V 60Hz.
-  CT is ZMCT116-A (2000:1)
- PT is ZMPT101B (1000:1000)

If pdf doesn't work, here is a link to images of my schematic as well: https://imgur.com/gallery/GHqTz9C

Thanks again.

[M90E36A.pdf](https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/files/5898876/M90E36A.pdf)

TSMotter, 2021-01-31:
Hello,
I just felt like writing this.... I understand that I opened an issue on your repo with something that is not particularly related to your project and I'm sorry about that...

It's just that I'm trying to gather every piece of info I can get on these IC's on the internet with people who have already used it before, since Microchip documentation is poor and confusing.

If you could perhaps share any resource (hardware/firmware) about this side project you mentioned using the 36, you'd still be helping me **a lot** and it'll probably be a much simpler process then the other way around, where I be asking you to review my Hw/Fw...

Anyways... Thank you again and sorry for the inconvenience.

CircuitSetup, 2021-02-04:
No problem! Your schematic looks okay to me. 

It sounds like SPI is communicating properly with the ATM90E36 since you're getting a number just below 65535. 

What do you have set for the voltage gain?  

Also, I recently updated the Arduino library for the 36 here: https://github.com/CircuitSetup/ATM90E36  
See what was changed from the original: https://github.com/CircuitSetup/ATM90E36/commit/6e3c9c6cc3c41f6518bee7a50c7359a5e861296f#diff-732495b4c2138fbb3a8a74587967ffbd7d351025c2aec717dddbcd14ce04a5b3
Are you using this version?

TSMotter, 2021-02-06:
Thanks for sending the references bud...

> What do you have set for the voltage gain?

That's something I dont fully understand. How can you start with a value for the voltage gain, if it's value depends on user inputing referente meter reading. Currently what I do is that I start with the default voltage gain value (0xCE40), and then run a calibration process on the go, where I input reference meter reading value, my function makes a series of X (currently 20) readings of the ATM voltage value, takes the mean value and write to the voltage gain register the resulting value, according to the process sugested on AppNote section 4.2.6.

But as my voltage readings from the register are so off, I'm thinking I have a hardware issue.

I'll take a time to go thru your ATM90E36A repo, the simulation and the .xls file should help me a lot.

Thank you again.

closed 2022-03-25 by CircuitSetup