CircuitSetup/Split-Single-Phase-Energy-Meter #47: fried split circuit v1.4 rev 2

chaseadam, 2021-02-06:
Installation running for a bit over 4 months paired with a Lolin knockoff esp8266 reporting to local emon. It was in "open air" installation (sitting on shelf next to power panel).

Smelled the classic electrical burning smell after breakfast and found the power meter off and a bit "toasty" (not the breakfast kind).
[image]
[image]

I am using one of the preferred high quality power supplies
[image]

I haven't done much investigation yet, but will keep this thread up to date.

I suspect maybe the low quality dev board was pulling too much current and slowly damaged the regulator to the point of catastrophic failure.

Curious for others' hypotheses as well.

CircuitSetup, 2021-02-06:
From the looks of it, the rectifier and power supply failed on the meter. I've only seen this happen once before, and the cause was likely because of a component failing. 

If you send me your order number I can send a replacement.

chaseadam, 2021-02-06:
responded via https://circuitsetup.us/index.php/contact/

closed 2021-02-06 by chaseadam