CircuitSetup/Split-Single-Phase-Energy-Meter #50: Setting up Split Single Phase meter on emoncms.org

farzadb, 2021-04-16:
Hello. On the wiki it says:

`For all but the EmonCMS.org service, (currently for EmonCMS.org these feeds and inputs have to be setup manually) you can automatically setup the energy meter device in EmonCMS:`

And I'm wondering if there's any help or step by step guide for how to setup the feeds and inputs manually on emoncms.org?

Thanks.

farzadb, 2021-04-17:
emoncms.org doesn't allow custom devices. So I had to setup my own instance.

closed 2021-04-17 by farzadb

CircuitSetup, 2021-04-17:
> emoncms.org doesn't allow custom devices. So I had to setup my own instance.

Sorry I didn't get back to you sooner. 

You can use emoncms.org, but the inputs will have to be connected to the feeds manually. When EmonESP sends data to emoncms.org, it will show the abbriviated fields, as seen in this screen shot under Name: https://raw.githubusercontent.com/CircuitSetup/Split-Single-Phase-Energy-Meter/master/images/emoncms_device_inputs2.PNG

Those are what have to be connected to tell emoncms what to do with them.

farzadb, 2021-04-17:
No worries. Thanks for getting back to me. 

Ah I see. I'll play around with that and see if I can figure out how to connect the inputs to feeds. I'm also happy with my own instance of emoncms for now.