CircuitSetup/Split-Single-Phase-Energy-Meter #51: support for high frequency AC transformer?

asalexsu, 2021-05-02:
Hi,

I couldn't find any outlets near the electrical box so I bought a 120V to AC 12V transformer, something like this
https://www.aliexpress.com/item/1005001682049282.html?spm=a2g0s.9042311.0.0.52cb4c4dN5ulqY

When I was testing with a multimeter it doesn't read anything(shows as 0). The seller said I will need to get a reading with a high-frequency multimeter

In ESPhome the voltage reading varies between 1.8V to 2.3V with a gain of 29462

Does this mean it doesn't support reading from high-frequency transformers and I will need to hard code the voltage for other uses?

Thanks,
[image]

CircuitSetup, 2021-05-02:
Some questions: 
- Is the transformer hooked up directly to a breaker in your panel? 
- Is so, is it a single pole, and is is connected to the breaker and neutral bus bar?
- How are you measuring the output of the transformer? Is the multimeter set to AC?

asalexsu, 2021-05-02:
> Some questions:
> 
> * Is the transformer hooked up directly to a breaker in your panel?
> * Is so, is it a single pole, and is is connected to the breaker and neutral bus bar?
> * How are you measuring the output of the transformer? Is the multimeter set to AC?

- Yes, the transformer is directly hooked up to a breaker in the panel
- Yes, it is a North American single pole breaker, I will attach a picture below
- I measured it by setting the multimeter to AC 200V, checked that the input reads 124.8V, output reads 0.3V

The transformer in black is the one powering the circuit board, the bulky one below is for security system

[image]

CircuitSetup, 2021-05-02:
Everything looks like it is hooked up properly. I would bet the transformer is defective if it's not outputting at least what it says it should. I'm surprised it's able to power the meter at all.

asalexsu, 2021-05-07:
I had some more time today and did some testing. The transformer seems to be using a high switching frequency. My multimeter reports in the 24k Hz - 25k Hz range as oppose to 60 hz.

Do you know if I can simply change the frequency setting in esphome to that number or would I have to get another transformer? Do you have any recommendation on 120v - 9/12V transformer that directly connects to the circuit?

Thanks,
[image]

CircuitSetup, 2021-05-08:
Something isn't right with that transformer - I highly doubt the 25kHz is coming from it. Being that high of a frequency is more likely interference from something else. You can't get a high frequency output from a transformer without some additional electronics. 

Either way, it doesn't seem like you'll be able to use that transformer. It's either broken or there is something else that I'm missing. I would suggest buying [this one and just taking it out of the case.](https://circuitsetup.us/index.php/product/jameco-ac-to-ac-wall-adapter-transformer-9-volt-1000ma-black-straight-2-5mm-female-plug/). Then you can cut off the plug prongs and wire to a breaker, just like your current one.