CircuitSetup/Split-Single-Phase-Energy-Meter #52: No AC adapter available

eavyon, 2021-05-28:
Because of EU regulations I'm told that AC/AC adapters aren't allowed to sell here anymore. That leaves me to using an AC transformer.

Is there any difference between an transformer and a nicely made adapter? Can I safely connect a transformer to the board or do I need to add some protection?

I've already ordered 2 sets for solar monitoring so I hope I can get it to work and post results.

CircuitSetup, 2021-05-28:
Any AC to AC adapter is the same as an AC transformer. The point being that you're taking an AC voltage and bringing it down to a much lower level. For the EU, this is usually 220-240V. The transformer needs to bring down the voltage to between 9 and 12VAC. 

As long as the transformer you found does this, it should work fine though. It will need to be calibrated to the meter according to the [instructions here.](https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter#voltage-procedure)

eavyon, 2021-05-28:
I scavenged a normal AD adapter and removed the rectifying part. It's measuring about 14v open circuit. I'll update when I have the set tested and calibrated.

CircuitSetup, 2021-05-30:
Are there still a bunch of capacitors in there? If so, this will not work very well for reading the AC signal, and could be very inaccurate, depending on how much the caps are smoothing the signal. It doesn't sound very safe either. I would recommend just finding a regular AC transformer.

closed 2021-06-03 by eavyon