CircuitSetup/Split-Single-Phase-Energy-Meter #54: Any advice for an european transformer brand?

wouldgo, 2021-07-17:
Hi,
I purchased yesterday the circuit from tindie and I was wandering stores and ebay searching an european plug (CEE 7/4, CEE 7/16 or  CEI 23-16 from [here](https://en.wikipedia.org/wiki/AC_power_plugs_and_sockets)) transformer but I then thought that maybe there are some suggested brands already.

Is there anyone that already had an installation with a transformer of that kind?
Thanks a lot

Dario

CircuitSetup, 2021-08-02:
My apologies for the delay. There are some recommended [AC power transformers with european plugs listed here](https://learn.openenergymonitor.org/electricity-monitoring/voltage-sensing/different-acac-power-adapters).

In general, anything that brings down the voltage to between 9 and 12 VAC and outputs 1000mA will work with the meter.

wouldgo, 2021-08-02:
Hi,

no problems at all. At the moment I've not yet found anything but after your advices I maybe found something with the rigth plug: as far I undestand [these](https://www.ebay.it/itm/264529555139?hash=item3d9730aec3:g:T1AAAOSwCtxfN7XL) specs are fitting the requirements. 

Am I rigth?

Thanks again
Dario

wouldgo, 2021-08-30:
As reference that transformer is 100% fitting for the need.

closed 2021-08-30 by wouldgo