CircuitSetup/Split-Single-Phase-Energy-Meter #66: Alternative way to power the v1.4 board

foureight84, 2022-05-11:
My electrical box is oddly placed externally and there's no outlet nearby to utilize the  Jameco 157041 wall adapter. Is there a safe alternative way to power this board and keep it enclosed in the breaker box? Sort of like the Sense power box that uses an internal breaker as a power source.

CircuitSetup, 2022-05-25:
My apologies for the delay in responding.

The easiest way to do this would be to wire the transformer directly to a breaker in your panel. You can open the transformer case, and replace the plug prongs with wires that go to the breaker. If you've never done anything like this before, I would recommend hiring a professional.

Technically you're not supposed to mix high and low voltage in a panel, so you may want to put the transformer and meter in a separate weatherproof box.

foureight84, 2022-05-29:
Thank you! That is very helpful. I didn't realize the Jameco 157041 could be disassembled.

CircuitSetup, 2022-06-01:
Yep, you just have to break the glue on the seams.

closed 2022-06-01 by CircuitSetup