CircuitSetup/Split-Single-Phase-Energy-Meter #68: Rebooting problem using emonESP with ESP32 with Arduino IDE

albnar, 2022-05-24:
After uploading the src.ino code from emonESP into a Espressif ESP32-Wroom-32 board using Arduino IDE, the ESP32 is reloading continuosly.  Before loading the code it broadcast its WiFi name, but after uploading the code it disappered and reboots continuosly.
Here is the text in serial monitor:

16:34:55.384 -> Rebooting...
16:34:55.384 -> ets Jun  8 2016 00:22:57
16:34:55.384 -> 
16:34:55.384 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:34:55.384 -> configsip: 0, SPIWP:0xee
16:34:55.384 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:34:55.384 -> mode:DIO, clock div:1
16:34:55.443 -> load:0x3fff0018,len:4
16:34:55.443 -> load:0x3fff001c,len:1216
16:34:55.443 -> ho 0 tail 12 room 4
16:34:55.443 -> load:0x40078000,len:10944
16:34:55.443 -> load:0x40080400,len:6388
16:34:55.443 -> entry 0x400806b4
16:34:57.271 -> Starting AP
16:34:57.365 -> WiFi interface ready
16:34:57.365 -> WiFi client started

I tried with three new brand boards.  The problem appears in every one of the boards.
In the instructions this issue ist mencioned and a total erase of RAM flash is recommended.  It did not help.  

I would appreciate any help.
Best regards

CircuitSetup, 2022-05-25:
Try [disabling the watchdog timer here.](https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/blob/e1790d1bc7859581c94a0c7407c2ae6370ee8dd2/Software/EmonESP/src/emonesp.h#L38)

albnar, 2022-05-25:
Hi John,
thank you very much for your prompt reply.
Your recommendation was very helpful.
Now I get the WiFi Name and I was able to connect to him.  If I try to
navigate  to 192.168.4.1 I get this:  I am using my Laptop to connect WiFi.

/home.html not found, have you flashed the SPIFFS?

I can not go further:

[image: image.png]

I read the Issue #69: SPIFFS log error en version 3.0.0 #69. I compiled
with arduino IDE. It seems to be for Phyton.
Sorry, but I still need your help.  I do not know what to do with this.
 What have I to do?

Best regards,

Alberto

albnar, 2022-05-25:
Hi John,
maybe this information is important for you:
[image: image.png]

albnar, 2022-05-27:
Hi John,
I still try to get this application running.
During compiling in verbose mode option I see following text regarding
SPIFFS
~~~
Compiling library "SPIFFS"
"C:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++"
-DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\""
-DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bt"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/coap"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/console"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/driver"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/efuse"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-tls"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp32"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_adc_cal"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_event"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_client"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_server"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_ota"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_server"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_ringbuf"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_websocket_client"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/espcoredump"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/ethernet"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/expat"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/fatfs"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freemodbus"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freertos"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/heap"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/idf_test"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/jsmn"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/json"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/libsodium"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/log"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/lwip"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mbedtls"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mdns"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/micro-ecc"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mqtt"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/newlib"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/nghttp"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/nvs_flash"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/openssl"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/protobuf-c"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/protocomm"
"-IC:\\Users\\ANC\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/pthread"
[... cut ...]
>>> <https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/blob/e1790d1bc7859581c94a0c7407c2ae6370ee8dd2/Software/EmonESP/src/emonesp.h#L38>
>>>
>>> —
>>> Reply to this email directly, view it on GitHub
>>> <https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/issues/68#issuecomment-1137470410>,
>>> or unsubscribe
>>> <https://github.com/notifications/unsubscribe-auth/AIKGYC23ZTY42YXSWU27RW3VLZEUDANCNFSM5W3AZIZA>
>>> .
>>> You are receiving this because you authored the thread.Message ID:
>>> <CircuitSetup/Split-Single-Phase-Energy-Meter/issues/68/1137470410@
>>> github.com>
>>>
>>
>>
>> --
>>
>>
>>
>
> --
>
>
>

--

albnar, 2022-05-27:
by the way
in serial monitor I see following textevery second:
14:34:36.009 -> Sys Status: S0:0x1770 S1:0xfa00
14:34:36.009 -> Meter Status: E0:0x370 E1:0x7200
14:34:36.009 -> .
14:34:37.003 -> Sys Status: S0:0x1770 S1:0xfa00
14:34:37.003 -> Meter Status: E0:0x370 E1:0x7200
14:34:37.003 -> .

albnar, 2022-05-27:
Problem solved following this case:

Adafruit ESP32 Feather hangs at init when calling SPIFFS.begin() #888
best regards.
Alberto

CircuitSetup, 2022-06-01:
Glad you were able to figure this out!

closed 2022-06-01 by CircuitSetup