CircuitSetup/Split-Single-Phase-Energy-Meter #73: Getting 65535 values for Current and voltage

abraralishah12, 2022-09-27:
I am currently working on a PROJECT and ordered your PCB without esp32 full kit. And I am getting 65535 values for both current and voltage

CircuitSetup, 2022-09-27:
The first thing to check is that everything is wired correctly. 
Do the meter and esp32 share a ground? 
Are the SPI wires connected on the pins specified in the program?
Also, how is everything being powered?

abraralishah12, 2022-09-27:
Yes they share a ground. SPI wires are connected as specified in the program. Currently, I am powering them from USB port. I haven't connected the 9V transformer yet. I used a 9V transformer earlier but it had damaged my IC and ESp32. So right now I am only checking Current value.

abraralishah12, 2022-09-29:
@CircuitSetup kindly assist me on above issue.

CircuitSetup, 2022-09-29:
Wait, what exactly did the 9V AC transformer burn out? It should only be plugged directly into the meter. 

When you power the ESP32 via micro USB, is the meter showing a green light? Do any of the other lights blink when a current transformer is hooked up and around a wire with current going through it?

abraralishah12, 2022-09-30:
When I power esp32 with USB, green led blinks. When I ordered the PCB resistor array was short so other LEDs did not blink because of the open path in the PCB circuit.

abraralishah12, 2022-09-30:
@CircuitSetup can I get your email or WhatsApp number?

abraralishah12, 2022-10-04:
@CircuitSetup kindly have a look at it.

CircuitSetup, 2022-10-04:
> When I power esp32 with USB, green led blinks. When I ordered the PCB resistor array was short so other LEDs did not blink because of the open path in the PCB circuit.

Is there a solid green light on the meter?

Specifically, what program are you running on the ESP32?

abraralishah12, 2022-10-05:
yes, green led, and it blinks when I power it with USB. 
Below is the code:
`

#include <ATM90E32.h>

const int CS_pin = 5;
/*
  18 - CLK
  19 - MISO
  23 - MOSI
*/

unsigned short VoltageGain = 37106;
unsigned short CurrentGainCT1 = 25498;
unsigned short CurrentGainCT2 = 25498;
unsigned short LineFreq = 4485;
unsigned short PGAGain = 21;

ATM90E32 eic{}; //initialize the IC class

void setup() {
  /* Initialize the serial port to host */
  Serial.begin(115200);

  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB
  }

/*
  Wire.begin();

  display.begin(SSD1306_SWITCHCAPVCC);

  // Clear the buffer.
  display.clearDisplay();
  display.setCursor(0, 0);

  display.setTextSize(1.5);
  display.setTextColor(WHITE);
*/

  /*Initialise the ATM90E32 + SPI port */
  Serial.println("Start ATM90E32");
  eic.begin(CS_pin, LineFreq, PGAGain, VoltageGain, CurrentGainCT1, 0, CurrentGainCT2);
  delay(1000);
}

void loop() {

  /*Repeatedly fetch some values from the ATM90E32 */
  float voltageA, voltageC, totalVoltage, currentA, currentC, totalCurrent, realPower, powerFactor, temp, freq;

  voltageA = eic.GetLineVoltageA();
  // Voltage B is not used
  voltageC = eic.GetLineVoltageC();
  totalVoltage = voltageA + voltageC ;
  currentA = eic.GetLineCurrentA();
  // Current B is not used
  currentC = eic.GetLineCurrentC();
  totalCurrent = currentA + currentC;
  realPower = eic.GetTotalActivePower();
  powerFactor = eic.GetTotalPowerFactor();
  temp = eic.GetTemperature();
  freq = eic.GetFrequency();

  Serial.println("VA:" + String(voltageA) + "V");
  Serial.println("VC:" + String(voltageC) + "V");
  Serial.println("IA:" + String(currentA) + "A");
  Serial.println("IC:" + String(currentC) + "A");

  delay(1000);
  delay(14000);

/*
  //Clear display to prevent burn in
  display.clearDisplay();
  display.display();
*/
}
`

abraralishah12, 2022-10-07:
@CircuitSetup KIndly have a look at it.

abraralishah12, 2022-10-08:
@CircuitSetup please have a look at it

abraralishah12, 2022-10-10:
@CircuitSetup have a look at the code please