CircuitSetup/Split-Single-Phase-Energy-Meter #75: Energy meter solar kit installation

MassimilianoBiancucci, 2022-11-18:
Hi, 

I have a few doubts related to the installation of the solar kit with 2 energy meter boards.
I would like to install the solar kit to measure the energy consumed by two separate lines, the grid and a solar inverter output, both at 230V 50HZ, in this configuration, the grid line is hot at every moment, while the inverter line can be turned off periodically.

Reading the section "Plug in the AC Transformer" I had a few doubts :
(Sorry if some question seems dumb I'm not so familiar with the terms used in that section in English)
I don't get how the energy meter measures the grid voltage, Is the voltage read directly from the 9V AC power transformer?
And the 9V AC transformer must be connected to the same line as the CTs connected to the relative board?

So if my assumption is correct, if I want to read the values (Volts, Amps) from one line I need to connect the 9V AC transformer to the same line, for example, if I want to measure the power from the grid with the first energy meter board I have to connect its supply transformer to that one, and if with the second energy meter board I want to read the values from the solar inverter output, the second energy meter's  9V AC transformer should be connected to that line? Am I correct?

considering this configuration if the inverter line is turned off and one of the two energy meter boards in the solar kit is turned off (due to the absence of supply) can be a problem for the circuit?

I added an image with the configuration that I tried to explain, is that wrong? 

[image]

NoNameSwede, 2022-11-28:
I have a simular question. Have 230V 50Hz but three phases. Done the configuration but can´t get the voltage calibration right.

# Jameco 9VAC Transformer: 
#  For meter versions: 
#  >= v1.3: 7305
  voltage_cal: '7305'

This setup is for 120V transformer. Around what value should be used for 230V transformers? When I turn this all the way up to 65535 I get a reading around 13V i Home Assistant. And if I use 7305 I get around 0,3V.

NoNameSwede, 2022-11-28:
Hi

Deleted my post when I realized this was for the the 2 phase version. I have the 6 channel board. The question remains though.