CircuitSetup/Split-Single-Phase-Energy-Meter #77: Installation help

k7faq, 2023-02-01:
Where can I find instructions to follow to install the software needed to configure these devices I purchased?

Attempting to follow https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter/tree/master/Software/EmonESP#option-1-using-platformio
- I am lost in what is required to use the IDE. It does not show me any devices that report as ESP32 or like when I list my devices.
- Following the download instructions for the CLI my system fails to find `pio`. 

I also tried using Arduino IDE. Again, I could not figure out what software was required where to cause Arduino IDE to recognize my devices.

I am new to using Arduino products and really just do not have the time to go on a long journey to figure this out. 

Can someone help me?
I am using Mac OS 13.1 with Python 3.10.6

CircuitSetup, 2023-02-03:
I'm sorry the process is so confusing. Are you installing on an ESP32 for the single phase or 6 channel meter? Do you want to use EmonESP or ESPHome? EmonESP is best used for EmonCMS, where ESPHome is best used with Home Assistant.

k7faq, 2023-02-03:
I have two 6 channel boards. I am not sure which EmonESP or ESPHome would best suit my needs so I could try each.
Reading the docs I am confused on the process in which to be able to reach a point I can download the application to the ESP32 board.
Trying Platform.io I encounter these errors:
`In file included from src/mqtt.cpp:30:
src/mqtt.h:36:10: fatal error: AsyncMqttClient.h: No such file or directory`

I managed to add this library and that error is gone. But I am stumped with this error that I do not understand which library to be installed to resolve.

`src/emoncms.cpp: In function 'void emoncms_publish(const char*)':
src/emoncms.cpp:49:5: error: 'WiFi' was not declared in this scope`

k7faq, 2023-02-03:
When I opened the wifi.cpp file I find that "WiFi is undefined" on line 92 `WiFi.enableAP(true);`. Same holds true for other instances of "WiFi'.