CircuitSetup/Split-Single-Phase-Energy-Meter #78: one channel CT channel reading differently

chaseadam, 2023-06-26:
I am using the example ESPhome config and noticed that one CT channel reacted very differently to my large (supposedly) balanced 240V EVSE load. Fhase C reacts, but with much lower absolute reading.

I swapped the CT inputs and I got the similar channel readings which should eliminate the CT or the load as a cause.

I switched in a v1.4 rev3 board and am getting similar magnitude of readings from both channels (but with very off calibration).

Now that I bothered to type all this out, I think my issue might be that I failed to sufficiently sever the burden resistor jumper on CT2 channel because my CTs have built in burden resistors. Will report back after investigation.

MaiaBlanco, 2023-07-06:
I have spent considerable time trying to debug this issue myself and do not have a solution yet. I suspect a software issue may be at play, because I am able to calibrate the CTs I am using to reflect current fairly accurately. 

The CTs I am using are 200A/50mA Current Transformer YHDC SCT-024 – 24mm Opening as sold on CircuitSetup's website [here](https://circuitsetup.us/product/200a-50ma-current-transformer-yhdc-sct-024-24mm/?v=7516fd43adaa). Per your concern about burden resistors, I am not aware that these CTs have them (it is not mentioned in the documentation nor on CircuitSetup's storefront). For reference, the current calibration value I ended up computing based on Kill-a-Watt measurements with a resistive load is 49712. 

Similar to the original post, I am seeing that one of the phases has a much lower computed Watts reading even though the current scales roughly as expected when I apply a resistive load. I can swap the CT inputs on the meter board and have the same behavior reproduce on the other channel. And I previously swapped the two CTs on the mains lines and _think_ the behavior didn't follow the CT; it seems to follow the phase in the breaker box. 

I'm not sure if this is a software problem or perhaps an issue with my installation. OP, please reply back if you discover that burden resistors were the culprit!

CircuitSetup, 2023-07-07:
@chaseadam were you able to resolve this by severing the burden resistor?

@MaiaBlanco can you confirm that the CTs are [installed on your mains like this](https://github.com/CircuitSetup/Split-Single-Phase-Energy-Meter#connect-current-transformers-to-your-mains)?

MaiaBlanco, 2023-07-08:
I confirmed that my current clamps are installed on the mains as shown, yes. The first channel (CT1) is in stalled on the hot line that feeds the same breaker that the voltage monitoring is connected to. The direction (arrow) on the clamp points into the breakers. The second current clamp (CT2) is on the other hot line, pointing opposite to the first. 

I think I had a reversed setup before, where the directions of both clamps were reversed from the above, because now I observe negative values reported in the interface for power factor and voltage. Reactive power is being reported as a positive value. 

My problem still persists: on the second phase which is monitored by CT2, when I apply a water kettle as a load, the reported wattage is significantly lower than it actually is. For reference, I have checked and recalibrated both clamps individually with a clamp-style multimeter on each hot line, and the reported current values are correct. It's the wattage that is wrong, and only for the CT2 (second phase) channel. The magnitude of the discrepancy is almost 2x: I see an increase of ~512 watts (magnitude, since the values are being reported negative now) through CT2 while I see ~950 watts in the kill-a-watt meter I have on the kettle itself. There are no other loads turning on or off while I run this tests.

CircuitSetup, 2023-07-08:
@MaiaBlanco If you're seeing negative values anywhere that usually means something is reversed. This is going to sound silly, but turn around the voltage transformer in the socket. 

I bet you're seeing low wattage values because the meter is calculating power factor out of phase. 

If this doesn't fix the issue, please supply your config file.

MaiaBlanco, 2023-07-08:
That makes sense wrt to negative values! Unfortunately, while the wattage values are no longer negative, the second current sensor is still far off on absolute wattage. I also note that while the power factor on the first current meter reaches 1 when I apply a large resistive load (and correctly reports wattage), the second current channel tops out at a power factor of 0.5 and reports roughly half of the wattage it should, as o described earlier.

MaiaBlanco, 2023-07-08:
Here is my config (minus wifi and other ESPhome boilerplate:
`substitutions:
# Change the disp_name to something you want
  disp_name: EMON
# Interval of how often the power is updated
  update_time: 1s

web_server:
  port: 80

spi:
  clk_pin: 18
  miso_pin: 19
  mosi_pin: 23

sensor:
  - platform: wifi_signal
    name: ${disp_name} WiFi Signal
    update_interval: 60s
  - platform: atm90e32
    cs_pin: 5
    current_phases: 2
    phase_a:
      voltage:
        name: ${disp_name} Volts
        accuracy_decimals: 1
      current:
        name: ${disp_name} CT1 Amps
        id: "ct1Amps"
      power:
        name: ${disp_name} CT1 Watts
        accuracy_decimals: 1
        id: "ct1Watts"
        # filters:
# I don't know why but this one is always really off, even though the current measures correctly??
          # - multiply: 1.173
      reactive_power:
        name: "Reactive Power CT1"
        # filters:
        # I don't know why but this one is always really off, even though the current measures correctly??
        #   - multiply: 1.173
      power_factor:
        name: "Power Factor CT1"
        # filters:
        # I don't know why but this one is always really off, even though the current measures correctly??
        #   - multiply: 1.173
# Voltage using a 9VAC power transformer
      gain_voltage: 3920
      gain_ct: 51074
    phase_c:
      current:
        name: ${disp_name} CT2 Amps
        id: "ct2Amps"
      power:
        name: ${disp_name} CT2 Watts
        accuracy_decimals: 1
        id: "ct2Watts"
# If the wattage is backwards due to the phase 180 degrees from the clamp direction
        # filters:
        #   - multiply: 1.0
      reactive_power:
        name: "Reactive Power CT2"
        # filters:
        #   - multiply: -1.0
      power_factor:
        name: "Power Factor CT2"
      gain_voltage: 3920
      gain_ct: 48845
    frequency:
      name: ${disp_name} Freq
    line_frequency: 60Hz
    gain_pga: 2X
    update_interval: ${update_time}
  - platform: template
    name: ${disp_name} Total Amps
    id: "totalAmps"
    lambda: return id(ct1Amps).state + id(ct2Amps).state;
    accuracy_decimals: 2
    unit_of_measurement: A
    update_interval: ${update_time}
    icon: "mdi:flash-circle"
  - platform: template
    name: ${disp_name} Total Watts
    id: "totalWatts"
    lambda: return id(ct1Watts).state + id(ct2Watts).state;
    accuracy_decimals: 0
    unit_of_measurement: W
    icon: "mdi:flash-circle"
    update_interval: ${update_time}

  - platform: total_daily_energy
    name: ${disp_name} Total kWh
    power_id: totalWatts
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh

time:
  - platform: sntp
    id: sntp_time

switch:
  - platform: restart
    name: ${disp_name} Restart`

CircuitSetup, 2023-07-08:
Okay, so everything is positive now? If so, what happens when you switch the CT inputs on the meter, and turn around the voltage transformer at the same time? Is CT1 now reading too low? If so, there is something up with the CT - it's either broken somehow, or is too loose/not closed all the way on the wire.

If not, then there is something wrong with the meter (likely a cold solder joint), and I'll send you a replacement.

I don't see anything wrong with your config, btw.

MaiaBlanco, 2023-07-08:
I switched the CT inputs on the meter, turned the voltage meter around, and got both negative wattage values AND incorrect wattage readings on CT1 (where previously they were on channel 2)

MaiaBlanco, 2023-07-08:
But, I’m confused, because the current values are correct. It’s just all the other values computed that seem incorrect. I know this because I calibrated both CTs independently using a resistive load and a current clamp multimeter. How can the current values be reported accurately but the wattage is consistently wrong for one of the current clamps?

MaiaBlanco, 2023-07-08:
Could the calculation be wrong because it’s the second phase, and not the clamp? I’ll try swapping the clamps to the other phases and see if the issue follows the phase in the breaker box or if it follows the clamp.

MaiaBlanco, 2023-07-09:
Update: I have swapped the current clamps between mains lines and between channels, such that the same channel on the PCB is monitoring the same mains line but with the other current clamp. I still see the discrepancy in which the second current monitoring channel under-reports wattage, but NOT current. This is starting to seem more and more like a software bug. Any thoughts?

CircuitSetup, 2023-07-09:
> Update: I have swapped the current clamps between mains lines and between channels, such that the same channel on the PCB is monitoring the same mains line but with the other current clamp. I still [...]

So the issue followed the meter channel, not the phase in your box, correct? If that's the case, I'll send a new meter for you to test. Can you provide your order number?

MaiaBlanco, 2023-07-09:
> So the issue followed the meter channel, not the phase in your box, correct? If that's the case, I'll send a new meter for you to test. Can you provide your order number?

No, the issue seems to have followed the phase in the box. It does not follow the meter channel, nor does it follow the current transformer clamp. The only factor I have not tested independently is putting the voltage measurement on the same phase that is giving bad wattage but not current readings. I will try this now.

MaiaBlanco, 2023-07-09:
So it seems that whichever phase does NOT have the voltage reading, does not report watttage correctly. It doesn’t matter which current clamp or which channel on the meter board is monitoring which phase. It’s just, whichever phase is not the same one the voltage transformer is connected to, this phase gives incorrect wattage readings.

MaiaBlanco, 2023-07-09:
Perhaps there is something wrong with the board, but it’s not the current sensing but the voltage sensing for channel 2? (Note: my configuration has just one voltage input for both phases).

CircuitSetup, 2023-07-09:
Ohh okay! 

So, the thing that is bringing down the wattage for CT2 (or phase 2), is the power factor. If the voltage were exactly 180 degrees out of phase with the 1st phase, and you only had resistive loads, the PF would be 1.

The calculations for wattage, (V * I) * PF, is done on the metering chip, and cannot be adjusted except by using the gain and calibration values. 

It's possible that something is wrong with a circuit on phase 2, or the transformer feeding your house is failing. Do your lights dim when a large load is turned on (like an air conditioner)? 

The meter, by default, reads 1 voltage equally, assuming there is nothing wrong with the resistors going into that voltage channel. In this case, a defect with the meter is still possible.

Turning the voltage transformer around on a phase 1 circuit will reverse the polarity 180 degrees. If phase 2 is still reading wrong after turning it around, then it is not 180 degrees out of phase with phase 1 as it should be. If you were to measure the voltage from a phase 2 circuit, then phase 1 wattage would likely appear incorrect (assuming there is nothing wrong with the meter like mentioned above).

CircuitSetup, 2023-07-09:
> the voltage sensing for channel 2? 

Thinking about this some more... what is the output of voltage 2 (phase c in the config) from the meter? If it's reading about the same as voltage 1 (phase a in the config), then the problem can't be with the meter.

MaiaBlanco, 2023-07-09:
I changed my settings to also dump the second voltage channel and it is indeed reading identically to the first. So this likely rules out the meter being defective, in addition to ruling out the current clamps as before.

I have a feeling you may be right; the two phases may not properly be out of phase (180 degrees apart). I’m going to have to figure out a way to measure this.

If I measure the AC voltage between the two hot lines, shouldn’t that give me a sense of their phase? If they are perfectly opposed I should see 120v x 2 peak to peak, and I should see less if the phase angle is anything else.

Also no, I do not observe lights flickering when a large load is applied. But I’ll keep in mind to investigate if anything else is wrong with my supply side.

MaiaBlanco, 2023-07-09:
Okay, I don’t recall if the multimeter will report peak to peak or RMS…probably RMS. In any case, I measure about 120v on each separate phase, and across both I see ~213v. Clearly not adding up.

CircuitSetup, 2023-07-09:
Thanks for checking! That does rule out the meter being defective.

> In any case, I measure about 120v on each separate phase, and across both I see ~213v.

This would make sense if you actually have 2 phases of a 3 phase wye configuration. Each phase is 120V out of phase from eachother. Are you in an apartment or single home?

MaiaBlanco, 2023-07-09:
It’s been a hot minute since EE coursework, but I think that I should be able to compute the relative phase offset of the two hot phases by using the RMS measurements I posted above.
`
123v+cos(theta) + 123v = 213.5v
1+cos(theta) = 1.73577
cos(theta) = 1.73577
theta = cos^-1(0.73577)
theta = 42.6 degrees (or 0.744 radians)
`

MaiaBlanco, 2023-07-09:
I am in an apartment, so maybe I am getting 2 of three phases. Did I measure something incorrectly or do the wrong math above?

MaiaBlanco, 2023-07-09:
So, how should the meter be configured when a home is getting 2 of three phases? Is a second voltage reading required or is there a configuration I can change?

CircuitSetup, 2023-07-10:
> I am in an apartment, so maybe I am getting 2 of three phases. Did I measure something incorrectly or do the wrong math above?
> 
> Edit: well a quick google search confirms that it’s probably three phase power and that I should be seeing approximately 1.73x the voltage across two phases versus measuring on one. Time to study back up…

Yep! That makes sense! 
> So, how should the meter be configured when a home is getting 2 of three phases? Is a second voltage reading required or is there a configuration I can change?

There are 2 solutions here:
1.  Use a 2nd voltage transformer to measure the 2nd phase & cut the voltage jumpers on the back of the meter
2.  In the config, calculate the watts using lambda, using the voltage and PF from phase 1 (this will be slightly less accurate, but still very close to actual): 
Make sure to give voltage and power factor an id:
`      voltage:
        name: ${disp_name} Volts
        id: "p1Volts"
        accuracy_decimals: 1
        unit_of_measurement: V
        device_class: voltage
`
`      power_factor:
        name: ${disp_name} CT1 Power Factor
        id: "ct1PF"
        device_class: power_factor
`

Then the template to calculate everything: 

`  - platform: template
    name: ${disp_name} CT2 Watts
    id: ct2Watts
    lambda: return (id(ct2Amps).state * id(p1Volts).state) * id(ct1PF).state;
    accuracy_decimals: 0
    unit_of_measurement: W
    device_class: power
    update_interval: ${update_time}  
`

MaiaBlanco, 2023-07-12:
Thanks for all your help debugging! I hope the original poster was able to resolve the initial issue. I will add a second voltage transformer to my setup and see how that goes.

chaseadam, 2023-07-12:
To close the loop for my original issue: I had scraped the jumper off, but I guess not well enough. It tested as shorted, so I re-scraped it and removed the burden resistor. Everything is back to normal for me.

@MaiaBlanco I appreciate the discussion as being related. Glad you got to the conclusion of the phase shift being the issue. I may consider another voltage transformer as well for more accuracy even though I have split phase.
@CircuitSetup as always, thanks for chiming in with the great detailed guidance.

Preview for the next issue: Debugging ESPHome on an ESP32C3-mini dev board that is locking up a few times per day at exactly 26 minutes and 21-28 seconds past the hour...

closed 2023-07-12 by chaseadam