CircuitSetup/Split-Single-Phase-Energy-Meter #81: 'WiFi' was not declared in this scope

RobertSchuster, 2024-12-26:
Compiling with Arduino IDE failes with _**'WiFi' was not declared in this scope**_

See attached file.
[compile Error.txt](https://github.com/user-attachments/files/18252511/compile.Error.txt)

Arduino IDE and all libraries are up2date.
Any ideas?

ptownhippie, 2025-01-25:
I read somewhere about changing something in the file concerning HTTP. Somewhere in the file HTTP needs to change to be changed to utilize Wi-Fi. I read this from a collaborator and I will try to find it and let you know.

CircuitSetup, 2025-05-29:
Please try using this build of EmonESP: https://github.com/CircuitSetup/EmonESP

bdokimakis, 2025-05-29:
I've installed all the libraries that are mentioned in https://github.com/CircuitSetup/EmonESP/blob/master/platformio.ini and could be found via the Library Manager of the Arduino IDE. Micro Debug@^0.0.4 was not among them. Only this https://github.com/rlogiacco/MicroDebug is available and doesn't resolve the problem I mentioned above.

Could you provide a thorough installation guide for Arduino IDE?